sebanc/brunch #529: wifi not working after dev 87 update

ameyjoshi3068, 2020-09-14:
I have inbuilt dell wireless and Bluetooth 1705 card. In dell Inspiron 3647 desktop.
I don't have issues previously But since update Version 87.0.4258.0 (Official Build) dev (64-bit) wifi isn't working.
I have many instability issues with this update but I can bear it.
here is dmesg: 
[dmesg.txt](https://github.com/sebanc/brunch/files/5215453/dmesg.txt)

ameyjoshi3068, 2020-09-14:
Now more serious problem arose
I am stuck before startup 
........Remounted ............ commit:600
I also tried to replace rootC with directions in #520 but it gave error no space to copy files

Luckily I already made backup of stable build and now I decided to restore from it

I am closing this issue now

sebanc, 2020-09-19:
The wifi issue is most likely due to a bug in the ath9k kernel driver which should be fixed in the next kernel version, for now I cannot downgrade the kernel to the previous version as chromeos does not like that (it triggers a powerwash).

My advice is to reinstall brunch r83 for now until brunch r86 is released (with a newer kernel).

ameyjoshi3068, 2020-09-19:
@sebanc  I noticed from previous issues that there is a brunch testing with 5.4 kernel
If you don't mind may I try testing it?
I am hopeful that I will be able to boot with it as I am able to boot with latest linuxmint 20

sebanc, 2020-09-19:
You should have received a github invite for "brunch-next" repo.

However, most Intel gen < 5 devices seems not to be able to boot it though (it stays on chromeos logo forever).

ameyjoshi3068, 2020-09-20:
Yes😟 I am unable to boot with it.
However I can boot ahed of chrome logo to complete black screen 
And it can load enough modules to turn off with power button, but nothing ahed seemed to have happened.

ameyjoshi3068, 2020-09-20:
I wish to contribute in this project but once my education got completed in a year or two.

ondr3j, 2020-09-21:
Any chance I could get an invite to the repo as well? I'm experimenting with the 5.4 kernel and would like to continue using brunch if I can

sebanc, 2020-09-21:
@ondr3j you have been added to "brunch-next" repo

aaufi, 2020-09-21:
Could you invite me as well? @sebanc I have an 8th gen Intel processor. Would love to try!

sebanc, 2020-09-21:
@aaufi you should have received the github mail invite

closed 2020-09-28 by ameyjoshi3068