sebanc/brunch #530: All bug on bytcr 5651 Intel Cherry Trail-T3 Z8300/Z8350

DevHamid, 2020-09-15:
I After several weeks using, I encounter bug, which is really annoying for me, hopefully we can figuring out this bug and also as feedback for development :.
1. Audio doesn't work, make playback error. 
2. Mic doesn't work.
3. Camera doesn't work.

I try brunch new version still not work, add disable_intel_hda and baytrail_chromebook still not work.
Added manually UCM bytcr-5651, audio only work on Wireless Bluetooth. 
At the same time, I try Linux mint, and work flawlessly without problem.

Thanks before. 🙏👍

sebanc, 2020-09-19:
ChromeOS is only compatible with usb cameras which is not the case usually on baytrail device, so there is not much I can do on that.

Regarding audio, both original chromeos r84 and r85 have a sound issue (fixed in the latest r85 stable update a few days ago), so only use r83 or latest r85 (after enabling updates) to test sound.

Then, could you try just running `alsactl -f /usr/share/alsa/ucm/bytcr-rt5651/asound.state restore` in crosh shell and check if you have sound ?

Perchemop, 2020-09-26:
I am having this same issue using Intel Atom x5-z8350.
Audio creates playback error only when connected and output is selected. If internal speaker is selected, playback works normally. Using r83.

I have tried running that command in crosh but it gives me the error unknown command: alsact1.
I am new to linux/terminal so apologise if this is an easy error.

I have not been able to add add disable_intel_hda and baytrail_chromebook to kernal as I can't figure out how to do this.

Hoping to solve this issue because I am otherwise really enjoying the experience with ChromeOS

sebanc, 2020-09-26:
@Perchemop after running the command, did you try to play audio ?

Perchemop, 2020-09-26:
> @Perchemop after running the command, did you try to play audio ?

@sebanc after running the command, playback skipping and freezing (when on the external audio selection) seems to be fixed but audio still has the same crackling/popping noises.

sebanc, 2020-09-26:
Could you try to replace the files in /usr/share/alsa/ucm/bytcr-rt5651/ with those ones ?
https://fedorapeople.org/~jwrdegoede/bytcr-rt5651/

sebanc, 2020-09-26:
or those config files also:
https://github.com/alsa-project/alsa-ucm-conf/tree/master/ucm2/bytcr-rt5651

DevHamid, 2021-11-27:
Hello, this bug still not resolved unfortunately. Last version audio works is Chrome os 83. Since that, never work anymore.

DevHamid, 2021-11-27:
Uploading VID_20211127_182553.mp4…

DevHamid, 2021-11-27:
https://user-images.githubusercontent.com/13690753/143677863-b1234dda-4e4e-4e77-9ad2-83ce94f2dd1b.mp4

DevHamid, 2021-11-27:
Update @sebanc already told me how to fix it on Telegram, just do this on shell
Try this one:
"sudo alsactl restore -f /usr/share/alsa/ucm/bytcr-rt5651/asound.state"