sebanc/brunch #533: Mouse scrolling not registered in Crostini.

Niels2468, 2020-09-19:
Hi,

I have been experiencing a bit of a problem with project brunch.
There is a bug that doesn't really register mouse scrolling except if scrolling a lot. It works fine in the browser (not crostini).
The problem is mostly experienced in games like Minecraft. Single scrolls are not registered. It only registers in bulk.
Any help? I tried wired and wireless.

Thanks!

sebanc, 2020-09-19:
I don't think it is a brunch issue but found this:
https://www.reddit.com/r/Crostini/comments/af0us9/scrolling_in_terminal/

It seems there are scrolling options / flags to setup

Niels2468, 2020-09-20:
Fixed after disabling the only checked option in mouse settings in the terminal options.

closed 2020-09-20 by Niels2468