sebanc/brunch #534: Best way on how to upgrade from release 81 to 85

jordanvanbergen, 2020-09-20:
Is it possible to upgrade from release 81 to 85 in 1 step?
And if yes how?
And if no what is the best way to upgrade from release 81 to 85?

sebanc, 2020-09-20:
I would suggest a clean install but otherwise you can:
1) enable updates
2) update brunch:
`sudo chromeos-update -f < path to the Brunch release archive >`
3) update chromeos

postadelmaga, 2024-10-05:
1. this command should be run where ? in chromeos shell ?  `sudo chromeos-update -f < path to the Brunch release archive >`
2. I cannot find anywhere How to update chromeos ? using UI or any shell command ?

I think it would be very useful to have a wiki section describing the update steps 

thanks