sebanc/brunch #535: Proper right click

TKFRvisionOfficial, 2020-09-21:
So for me everything worked beautifully. I have a Asus VivoBook Max D541N and used coral (i did this because the cpu build into my laptop was also used for a chrome book and that one use this). Because this is a normal laptop it has a proper right-click-button. But it dosent work. It just does the normal click. Ofcourse I can use two fingers but is there maybe a way to remap this button to the right click?

Thanks

sebanc, 2020-09-22:
Chromeos disables right click when multi-touch gestures are enabled, I don't think it is possible to have both.

TKFRvisionOfficial, 2020-09-22:
Thats to bad...anyways...can someone add the  "Asus VivoBook Max D541N" to the comaptible devices? Thanks...

closed 2020-09-22 by TKFRvisionOfficial