sebanc/brunch #538: not able to sleep with brunch r 85 testing 20200919

ameyjoshi3068, 2020-09-23:
Cannot sleep with latest brunch update by pressing shortcut search+shift+l
As soon as Chrome OS sleeps it turns on again in 1-2 seconds

I seen in changelog that the attempt to improve sleep has been implemented 

Can the fix be made optional (with framework option)?

Device: dell Inspiron 3647 pc
  With Intel 4th gen Intel(R) Pentium(R) Processor G3240 
Options: acpi_power_button , suspend_s3

ameyjoshi3068, 2020-09-25:
Can I downgrade brunch directly using `ChromeOS-update` or it will trigger powerwash?

sebanc, 2020-09-25:
I had no feedbacks confirming this change was actually helping with suspend so I am probably going to revert it in next release.

Going back to the previous brunch r85 should not trigger a powerwash as it is the same kernel version, however I cannot guarantee it and you should still backup your important data.

ameyjoshi3068, 2020-09-25:
OK Thanks

closed 2020-09-25 by ameyjoshi3068