sebanc/brunch #539: CrOS Update Server

Abhinavcs7, 2020-09-23:
Whenever I try to visit the page, it gives me this error: 500 Internal Server Error
The server has either erred or is incapable of performing the requested operation. I have tried accessing it from every device I have and it gives me the same error.

balopez83, 2020-09-23:
Please see closed issue #537 where I posted a link to an alternate site with the firmware files for download. It provides an easier to use interface for finding and downloading the version you are looking for. @sebanc  can you add the link I posted in issue #537 to your instructions since it seems Google may have pulled access to their site.

WesBosch, 2020-09-24:
They didn't pull access, the site goes down all the time. To prompt the page to load you can add cache:// to the beginning of the URL or use cros.tech (which pulls from the same sources) I do agree that these links should be mentioned as the cros-updates-serving site goes down pretty regularly

cache://cros-updates-serving.appspot.com/
https://cros.tech/

sebanc, 2020-09-25:
https://cros.tech/ has been added to the main readme