sebanc/brunch #540: Issue installing to MMC storage that was previously non-existent

5074pokemon, 2020-09-25:
I've noticed that I can no longer install the brunch framework from a USB flash drive. Older versions of the framework would install just fine, but I've recently encountered a trackpad issue, so I decided to re-image my asus netbook with a clean version of the latest framework and chrome recovery image, now the installer says that my storage is not a valid disk name despite what fdisk says.
[image]
[image]

sebanc, 2020-09-25:
I have indeed changed the destination device check in latest release, however it seems it behaves correctly for my sdcard.

Could you post the output of the 2 below commands ?

`if [ ! -b "/dev/mmcblk1" ]; then echo true; else echo false; fi`

`if [ ! "$(lsblk -npo pkname "/dev/mmcblk1" | grep /dev | uniq)" == "/dev/mmcblk1" ]; then echo true;
 else echo false; fi`

5074pokemon, 2020-09-25:
[image]

sebanc, 2020-09-25:
In theory both should be false. Could you also post the output of `lsblk -npo pkname "/dev/mmcblk1" | grep /dev | uniq` and also just `lsblk -npo pkname "/dev/mmcblk1"`

5074pokemon, 2020-09-25:
Both commands output nothing
[image]

sebanc, 2020-09-26:
Ok so this is the issue, could you post the output of `lsblk "/dev/mmcblk1"` ?

sebanc, 2020-09-26:
Nevermind, I will fix it in the next release by changing the way I check for valid block devices

5074pokemon, 2020-10-01:
Found a workaround to install onto my laptop, so I'm closing the issue.
Temporary Workaround:
Linux USB and two versions of the framework: The latest version, and a slightly older one that will still install. Borrow the install_chromeos.sh from the old version of the framework and use it to install Chrome OS directly to the flash storage with the new framework image files. So far, haven't found any issues with my installation.

closed 2020-10-01 by 5074pokemon