sebanc/brunch #542: MEDIACOM FlexBook_edge13-M-FBE13 - touchpad not detected

Lucaacer, 2020-09-25:
I am trying to use chrome OS on another device, a MEDIACOM FlexBook_edge13-M-FBE13.

I bought it a couple years ago to run a customized FydeOS, because I had found a way of removing the chinese stuff and converting it to an almost vanilla chrome OS (you still can find my old thread at xda-developers) but this little boy had a lot of issues with its touchpad.

There was a patch by Hans De Goede floating around which worked for a twin device by Trekstore, but did not work for my pc.

For some time I let my wife use this device with Windows 10 but, a month ago, I discovered that MX Linux, version 18.3, with this kernel 5.8.0-10.1-liquorix-amd64 #1 ZEN SMP PREEMPT liquorix 5.8-8.1~stretch (2020-09-17) x86_64 was fully supporting my touchpad.

Here follows everything about my current linux installation

+++ System Info
System         = MEDIACOM FlexBook_edge13-M-FBE13
BIOS           = HQ-BI-13.3-S133AR400-AA54S-013
Release        = MX 18.3 Continuum
Kernel         = 5.8.0-10.1-liquorix-amd64 #1 ZEN SMP PREEMPT liquorix 5.8-8.1~stretch (2020-09-17) x86_64
/proc/cmdline  = audit=0 intel_pstate=disable  BOOT_IMAGE=/boot/vmlinuz-5.8.0-10.1-liquorix-amd64 root=UUID=f40d5012-f134-4602-af38-c9a1d3e9d30e ro quiet splash
Init system    = sysvinit
Boot mode      = UEFI

Xinput list

Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ SYNA3602:00 0911:5288 Mouse             	id=9	[slave  pointer  (2)]
⎜   ↳ SYNA3602:00 0911:5288 Touchpad          	id=10	[slave  pointer  (2)]
⎜   ↳ Goodix Capacitive TouchScreen           	id=11	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Video Bus                               	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ USB 2.0 PC Camera: PC Camera            	id=8	[slave  keyboard (3)]
    ↳ Intel HID events                        	id=12	[slave  keyboard (3)]
    ↳ Intel HID 5 button array                	id=13	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]
    ↳ ACPI Virtual Keyboard Device            	id=15	[slave  keyboard (3)]

... and dmesg results

[    8.490721] input: SYNA3602:00 0911:5288 Mouse as /devices/pci0000:00/0000:00:16.2/i2c_designware.2/i2c-8/i2c-SYNA3602:00/0018:0911:5288.0001/input/input14
[    8.490982] input: SYNA3602:00 0911:5288 Touchpad as /devices/pci0000:00/0000:00:16.2/i2c_designware.2/i2c-8/i2c-SYNA3602:00/0018:0911:5288.0001/input/input15
[    8.491161] hid-generic 0018:0911:5288.0001: input,hidraw0: I2C HID v1.00 Mouse [SYNA3602:00 0911:5288] on i2c-SYNA3602:00
[    8.499153] uvcvideo 1-8:1.0: Entity type for entity Processing 2 was not initialized!
[    8.499158] uvcvideo 1-8:1.0: Entity type for entity Extension 6 was not initialized!
--
[    8.499650] USB Video Class driver (1.1.1)
[    8.529182] usbcore: registered new interface driver snd-usb-audio
[    8.566797] input: SYNA3602:00 0911:5288 Mouse as /devices/pci0000:00/0000:00:16.2/i2c_designware.2/i2c-8/i2c-SYNA3602:00/0018:0911:5288.0001/input/input17
[    8.567033] input: SYNA3602:00 0911:5288 Touchpad as /devices/pci0000:00/0000:00:16.2/i2c_designware.2/i2c-8/i2c-SYNA3602:00/0018:0911:5288.0001/input/input18
[    8.567339] hid-multitouch 0018:0911:5288.0001: input,hidraw0: I2C HID v1.00 Mouse [SYNA3602:00 0911:5288] on i2c-SYNA3602:00

Currently ubuntu 20.04 Lts doesn't support this touchpad out of the box, but I had some hope that your kernel could support it.

It doesn't, so I attach here the dmesg log.

I haven't even checked what else doesn't work, since the lack of a functional tochpad is a stopper.

Please, have a look at it and, if you can't directly integrate the patch into your kernel, tell me how to patch your kernel sources... I have a beefier device to compile on:)

sebanc, 2020-09-26:
Your FlexBook_edge13-M-FBE13 touchpad patch has been backported in the mainline kernel and is already enabled in the latest r85 release.

If you tried this release and touchpad does not work that's probably another issue, please try to see if the touchpad works with crouton as explained in #305

Lucaacer, 2020-09-26:
I tried the latest stable, i.e. 83 and something.

I will test the 85 testing and report, thanks for the suggestion.

Lucaacer, 2020-09-26:
Thanks, with the latest testing the touchpad works:)

closed 2020-09-26 by Lucaacer