sebanc/brunch #543: chromeos-install.sh - "/dev/nvme0n1 is not a valid disk name"

swebervna, 2020-09-25:
I installed Chrome OS with Brunch yesterday and it worked just fine, but after editing the grub I got it stuck in a boot loop so I wiped the disk and tried installing it again. But now, everytime I run the chromeos-install.sh script, I get the error "/dev/nvme0n1 is not a valid disk name", even though that is exactly what the disk is.

When I lsblk, I see /dev/nvme0n1. There's no way I'm typing it incorrectly.

Any help is greatly appreciated, thanks. :)

supechicken666, 2020-09-25:
Just type nvme0n1 during the script is ok

swebervna, 2020-09-25:
> Just type nvme0n1 during the script is ok

I tried that and I get:
df: nvme0n1: No such file or directory
Not enought space to create image file

supechicken666, 2020-09-25:
Try using gparted to make a new platform table

swebervna, 2020-09-25:
> Try using gparted to make a new platform table

I recreated my media, something with Brunch must've been damaged in my download from the website. It's working now 👍

closed 2020-09-25 by swebervna