sebanc/brunch #545: Chome OS won't wake from sleep

gjbnh1701, 2020-09-25:
Love this. The only problem I've run into is waking from sleep after being away for a while...solution? Otherwise, OUTSTANDING!

sbm7815, 2020-09-26:
I've had success by disabling TPM in my system BIOS. Both sleep and wake are now working for me. My one issue is that I will mainly be using this with Bluetooth connected devices and I noticed that I can't wake the system with a Bluetooth mouse or keyboard. It wakes fine with USB connected devices or by pressing the physical power button. Does anyone know a way to fix this? Outside of that, I'm finally all set. This is a fantastic project. Thank you guys so much!

sebanc, 2020-09-26:
@gjbnh1701 Indeed it seems disabling TPM helps with suspend. Which brunch release do you use ?

@sbm7815 If I am not mistaken the linux kernel does not allow bluetooth devices to wake the system by default, you would probably need to do something like this:
https://askubuntu.com/questions/848698/wake-up-from-suspend-using-wireless-usb-keyboard-or-mouse-for-any-linux-distro
If you find the /sys/bus/*/power/wakeup attribute that you need to change for resume from suspend to work with your device, I can help you setting a bootscript to enable it by default.

gjbnh1701, 2020-09-26:
Hi, thanks for your reply. I did some more research on this. Seems some
actual Chromebooks have problems waking from sleep mode, which suggests to
me the actual problem may reside in the ChromeOS. Anyways I found a temp
work around I can live with. There are a couple of Extensions which prevent
going into sleep...problem solved!!

sbm7815, 2020-09-28:
@sebanc 
Thanks, I reviewed the link. To my surprise, it was enabled both before sleep and after wake. I don't know why the Bluetooth devices weren't working, but after doing a lot of research, I suspected that it would likely be a hardware issue and nothing to do with CrOS. These were cheap Bluetooth devices from Amazon, so I went out and purchased a Logitech K580 keyboard and tried testing with this device. It worked, but inconsistently. Clearly, not all Bluetooth devices are made equal. I ended up switching over to the RF dongle, turned BT off and everything is seamless now.  

I reconfigured an old PC for an elderly person who finds computers extremely difficult to use. It's amazing to watch her efficiently use one now while talking to the google assistant. She is absolutely thrilled. Thanks again for your efforts.