sebanc/brunch #546: Tablet Mode (Manual Triggering) is Not Working

JosephInn, 2020-09-26:
Unfortunately, Tablet Mode (**Manual Triggering**) is Not Working for me. 
Hardware: **Surface Book** (Brunch r85 + ChromeOS 85)
Problem: The device switches back to desktop mode when the screen is **connected backwards**.

I Haven’t found any (**working**) way to manually change between desktop and tablet mode, **have tried**:
1. CTRL+ALT+SHIFT+T
2. “options=sysfs_tablet_mode” + “echo 1 | sudo tee /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode”
3. "options=force_tablet_mode"
4. Adding “--force-tablet-mode”  to “chrome_dev.conf”

This is the "sudo udevadm monitor" output:
[Description&Logs](https://docs.google.com/document/d/1TbYgcp4W0jKRHjsXvTw3OQ2gR-wLRZjUb28L6TgSnJk/edit?usp=sharing)

sebanc, 2020-09-26:
"The device switches back to desktop mode when the screen is connected backwards", what do you mean by that ?

Actually there are 2 tablet modes:
- with keyboard connected: same as clamshell mode but with an additional icon in the bottom right of the screen
- without keyboard connected: full tablet mode

Normally surface devices go in full tablet mode when the keyboard is disconnected.

JosephInn, 2020-09-27:
[--> Like This  <--](https://msegceporticoprodassets.blob.core.windows.net/asset-blobs/4017596_en_3)

I know of the two "tablet modes", but I think the first one isn't really useful. I would really like the "Full tablet mode" when the screen is backwards (like in the image), because in that position the keyboard and trackpad are behind the screen, inaccessible and deactivated, thus making it harder to use in Desktop Mode. This is the position I use the most when using ChromeOS, and I actually like the new **Tablet Gestures** and **Interface** (plus in Tablet Mode it allows the screen to **rotate just by rotating the device**, something that desktop mode doesn't do)

Also, when the screen is completely detached from the base, it does activate "full tablet mode". But I can't use it like that, because the battery cuts  down by 75% .

sebanc, 2020-09-27:
Ok, I get it now.

My suggestion would be to add "usbcore.surface_tablet_mode=0" on the kernel command line (to disable the built in tablet mode) and then you should be able to use CTRL+ALT+SHIFT+T to trigger tablet mode manually or  "options=force_tablet_mode"

JosephInn, 2020-09-27:
It seems to behave exactly the same after adding "usbcore.surface_tablet_mode=0", it did't seem to change anything (I added it after cros_debug), and using CTRL+ALT+SHIFT+T still triggers the "first" tablet mode instead of the "full" one.

JosephInn, 2020-09-27:
FOUND IT! 
All i had to do was add "**--force-tablet-mode=touch_view**" to "**chrome_dev.conf**"

Thanks man, couldn't have done it without your help.

closed 2020-09-27 by JosephInn

sebanc, 2020-09-27:
Ah that's nice, I did not knew about this flag.

Thanks for posting the solution :)