sebanc/brunch #547: Surface Go 2 Wifi not working

cdurau, 2020-09-26:
Hey guys,

I love project brunch and just bought a Surface Go 2 in the base version with 64GB to install it on. 
So far everything seems to work fine, even pen and bluetooth, except for the wifi connection. The toggle is off and if I turn it on, it won't find any Hotspots. The chip is Intel Wifi 6 AX200.

I tried the iwlwifi_backport option, but it did not work either. 

Are there any drivers or fixes for the Intel chip yet?

sebanc, 2020-09-27:
Hi,

The "iwlwifi_backport" option has been reported working on Surface Go 2. Did you install chromeos in dual boot or single boot ?

Could you post your grub config ?

Also,  is secure boot enabled on your SG2 ?

cdurau, 2020-09-27:
Hey Sebanc,

I added the framework options to the wrong line at the end of the grub.cfg not at the line starting with linux... Wifi is working now :-)

Thank you very much for your quick reply!

closed 2020-09-27 by cdurau