sebanc/brunch #549: Android app spinning in shelf and not starting randomly

johnnylau34, 2020-09-28:
After using Chrome OS for a while, after i tried to start Android app from app drawer or from the shelf, there is an app icon on the shelf which keeps on spinning and does not launch. This issue persists until I restarted Chrome OS. Before I restart the OS, I also noted that the touch keyboard no longer auto popups on touching a text field and the auto-rotation will not work until the OS is restarted too.

I am using Surface Go 2 LTE under Chrome OS 85 and Brunch Brunch r85 k4.19 20200919

[image]

sujal2102, 2020-09-28:
wait 2-3 minutes on login screen and then login and enter.

johnnylau34, 2020-09-28:
> wait 2-3 minutes on login screen and then login and enter.

Logout and login does work as a workaround. Sometimes though I am working on sth else and wouldn't want to logout...What could be causing this?

sebanc, 2020-09-29:
For now, I believe it is related to the recent chromeos cpu bug:
https://www.reddit.com/r/chromeos/comments/j0y6pk/new_google_chromeos_updates_cause_100_cpu_usage/

Crescendo-BLYAT, 2020-10-01:
@sebanc no bro, this issue is also on non-brunch chromebook, I have ASUS C214M with the exact same issue since v84.
Sometimes the whole Android subsystem dies or freeze randomly after you run certain Android app, in my case is **POST Mobile 4** app. 

Powerwash does nothing.

The quickest way to get it work is launching Linux terminal then issuing **adb reboot** rather than full reboot.
Good news is seems they partially fixed that in v85 (less frequent occurences).

Also my daughter's Acer ES1-432 Brunchbook with Coral image (now switched to Samus coz OS auto update never work in Coral image) is working perfectly since v83 till latest v85, never got any single issue. Strange eh?

Crescendo-BLYAT, 2020-10-01:
> > wait 2-3 minutes on login screen and then login and enter.
> 
> Logout and login does work as a workaround. Sometimes though I am working on sth else and wouldn't want to logout...What could be causing this?

Logout in Chromebook almost the same as rebooting, it close all subsystem.
In this case they shutdown the Android subsystem. Same when you issue **adb reboot** via Terminal (Linux subsystem).

Before that, please enable **Settings > Linux > Develop Android apps** then install adb in Linux subsystem.

koehnej86, 2020-10-27:
Ctrl+Alt+t
shell
sudo su
android-sh
stop
start
exit
This will restart the android subsystem. I had the same issue and found this trick online somewhere. Icons may still spin for 5-10 secs before the subsystem is initiated.

Crescendo-BLYAT, 2020-10-27:
> Ctrl+Alt+t
> shell
> sudo su
> android-sh
> stop
> start
> exit
> This will restart the android subsystem. I had the same issue and found this trick online somewhere. Icons may still spin for 5-10 secs before the subsystem is initiated.

same with adb reboot...
why the complicated way?

koehnej86, 2020-10-27:
For me, it was because I wasnt using a compatible recovery file with my Surface Pro 6. After trial and error, the nocturne recovery ended being the best to have android stable. Only thing I have to do is delete the ipu3_cio2 camera module dependencies to prevent the camera loop as stated in the forums. After that, everything works great, except for sleep, that's a known issue, but I just changed some settings in /user/share/power_manager and now it at least wakes from sleep with minimal issues.

PaddyMonahanLBNY, 2021-06-19:
Using a chromebook 4 from samsung with linux enabled, I was having this issue with okular. When I would open a PDF from the file folder, the app icon appears once in the system tray with the spinning wheel on top of it, and again underneath the main apps in the tray in the "recent" section, without the wheel spinning. 
When I open from the terminal I get 'Icon theme "CrosAdapta" not found' twice, then the app opens without the spinning wheel over it in the tray. When I open a PDF it does the same.