sebanc/brunch #550: MEDIACOM FlexBook_edge13 - touchscreen stops working after suspend

Lucaacer, 2020-09-28:
First of all, thanks for the recent brunch 85 testing, which made my touchpad work:)

Let's come to the rest, power management, brightness control and battery drain are fine, so the device can be used on a daily basis.

What does not work:

- when I fold the device, tablet mode does not get triggered and the hardware keyboard still accept inputs;

- the touchscreen does not work after suspend, please see here https://github.com/nexus511/gpd-ubuntu-packages/issues/10

Regarding the second issue, I see there is an available patch, has it been already backported or will it be included in the next releases?

It's not a request, just a question.

Thanks.

sebanc, 2020-09-29:
Regading touchscreen, I can try to backport goodix touchscreen drivers changes in next release.

Regarding tablet mode, that's a common issue, you will find your different options in #308

Lucaacer, 2020-09-29:
Thanks a lot.

I know the options about tablet mode,  but it's not that important for me, especially if the harware keyboard is still on (and I can't find a way to disable it).

sebanc, 2020-09-29:
In theory using "options=sysfs_tablet_mode" should disable your keyboard

Lucaacer, 2020-10-03:
> In theory using "options=sysfs_tablet_mode" should disable your keyboard

Yes, it just work as you said, thanks.

I just saved 2 scripts in /usr/bin and made them executable in order to switch between the 2 modes without typing 

echo 0 | sudo tee /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode

or

echo 1 | sudo tee /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode

which is not very easy the hardware keyboard has been disabled.

By the way, do you know a command line way to rotate the screen?

Lucaacer, 2020-10-14:
Thank you Sebanc, bug solved with the latest release!

Lucaacer, 2022-02-05:
Hello Sebanc, I reopened this thread because I am now using this device with your brioche chrooted Linux.

So, brioche doesn't work with the old 4.x kernel, while it works very well with all the other kernels you provided. On the other side, only the oldest kernel supports the touchscreen of my laptop and flagging the Goodix parameter doesn't do anything.

Please, have you got any solution? It looks that you backported goodix touchscreen drivers only for 4.x kernel.

Thanks.