sebanc/brunch #551: update to dev87

sujal2102, 2020-09-28:


sujal2102, 2020-09-28:
can i update chromeos to dev87 withoud any bug.

sujal2102, 2020-09-28:
I mean without

sujal2102, 2020-09-28:
what is brunch next repo

ameyjoshi3068, 2020-09-29:
@sujal it is not recommended to update to dev channel until matching brunch is available **but** if you want to test new features you surely can upgrade but be ready to face bugs(unlikely but can)
I would recommend a complete backup to be safe

about the brunch-next, it is a private repo for testing of higher brunch framework with 5.4 kernel
It is found that most devices with less than intel 4th gen processors are unable to boot with it.

sujal2102, 2020-09-29:
can you tell me which recovery images are for <=intel 3 processor

ameyjoshi3068, 2020-09-29:
This is mentioned in readme of this repo
Currently:

"rammus" is the recommended image for devices with 4th generation Intel CPU and newer.
"samus" is the recommended image for devices with 3rd generation Intel CPU and older.
"grunt" is the image to use if you have supported AMD hardware.

ameyjoshi3068, 2020-09-29:
I suggest you read the complete readme and then post issues

sujal2102, 2020-09-29:
thankyou very much but i was talking about other then samus.

sujal2102, 2020-09-29:
I have updated to dev 87 everything is excellent but camera not working. 
how to fix this ?

NSbuilder, 2020-10-14:
> @sujal it is not recommended to update to dev channel until matching brunch is available **but** if you want to test new features you surely can upgrade but be ready to face bugs(unlikely but can)
> I [...]

@sebanc Well since rammus is only recommended for intel 4th gen processors and up ( and also common processors ), would you please do a version of brunch that use kernel 5.4?

Also, since Chrome OS 86 binary is unavailable, whould it be ok to use Chrome OS 85 recovery with Brunch 86?

closed 2021-09-07 by sujal2102