sebanc/brunch #552: G Suite account login doesn't activate Google Play Store

mrtrivia, 2020-09-30:
I have the current version of brunch installed and running on a Dell 13 3000 2-in-1 (8th-gen i5, I think, 8GB of RAM, and a 1TB HD).

I installed brunch because Cloudready doesn't support Android phone sync or the Google Play Store.

I used the video you don't support (wry grin) as a basis for installing brunch. But I had to install the necessary packages by hand anyway. And since then I've read all your documentation. (I'm having issues with tablet mode, but I can just edit my GRUB configuration to take manual control of tablet mode--I did read the docs.) This is a better distro than Cloudready, and installing it was kind of fun.

My main Gmail account is a G Suite account. It doesn't end in "gmail.com". I originally got it so I can use Gmail at a custom domain. I'm the only user of this G Suite domain, so I'm also the superadministrator. All users on the domain have permission to access the Google Play Store. And with this account on my Android phone, I _can_ access the Google Play Store. (I can also change any other G Suite settings that are necessary.

When I first logged into brunch on this machine using my G Suite account, everything worked fine, but no Google Play Store. I tried everything I could think of, but no dice.

Then, I powerwashed the machine and logged in using an old, free Gmail account. Everything worked, and I had access to the Play Store (and more pinned apps showed up also).

I tried adding my G Suite account as a "person" under my free Gmail account, then restarting the machine That gives me access to the Google Play Store and Google Docs. But the Files app wouldn't access Google Drive under the secondary account (my G Suite account), and having to use the browser interface to Google Drive isn't going to cut it.

I tried it the other way, logging in with my G Suite account, adding my free account as a second "person," then restarting the machine. No difference--the Google Play Store didn't show up.

Is this an issue with brunch, with the G Suite configuration, or with Google?

If it's with Google, can you describe the issue for me? Since I pay them money, if all that needs to change is a setting behind the scenes, that may be possible.

Thanks for your considerable efforts so far, and for your help with this!

sebanc, 2020-09-30:
Brunch does not change anything related to the login process so I would guess the issue is related to your G Suite configuration but I don't use G suite so I cannot really help with that.

daemongloom, 2020-09-30:
That's how any Chromebook work. Guide to solve your issue:
https://medium.com/@dhd_llc/how-to-enable-google-play-store-for-g-suite-users-on-chromebook-3706dc884340

shivanshusingh2007, 2020-10-08:
It looks like your Gsuite administrator doesn't allow to use the account in play store
Check if you can use your Gsuite account in Play Store in your Android Phone

mrtrivia, 2020-10-13:
daemongloom, you did it!

The G Suite settings changed. But I asked G Suite technical support (which I should have done first!), and got an answer about 90 minutes later (in the evening).

To make a longer story shorter, I now have a fully operational Franken-Chromebook, for $0 spent, with specs that an $800 Chromebook couldn't match (i5, 8GB RAM, 1TB HDD).

Sebanc, if this keeps working as well as it has, I'm going to replace this laptop's 1TB HDD with a $70 Crucial 500GB SSD, and use it until it dies. THANKS!