sebanc/brunch #554: Stuck at ChromeOS LOGO

Gammre, 2020-09-30:
I was Install ChromeOS on a USB flash drive via this tutorial: https://github.com/sebanc/brunch#install-chromeos-from-windows. But when login ChromeOS, it stuck at ChromeOS LOGO! Please help!
I use Brunch r85 k4.19 testing 20200919 and Rammus R85.
My PC spec: 
Main B360
CPU I7-8700K
GPU 1070Ti
Ram 16GB

sebanc, 2020-09-30:
Chromeos will not work with nvidia gpu so make sure the intel integrated graphics card is enabled and that your display is plugged on it.

Adding "module_blacklist=nouveau" to the kernel command line might also help.

Gammre, 2020-09-30:
Many thanks!

dooloon, 2020-12-07:
> 非常感谢！

works or not

erwin-nasution, 2021-11-25:
Help me please!
I have pc using amd a4-5000 APU, ram 8gb and win 10.
When i booting on chrome os, then restart to menu boot, then i try again, same boot to menu booting, im install chrome os grant and zork. Help me please!
Thanks you.

IBRAHIM8655, 2022-05-13:
Help me i have aslo Nvidia gtx 650 after linux logo its goes to black screen