sebanc/brunch #555: Chrome OS becomes unbootable with unfinished system update

ameyjoshi3068, 2020-09-30:
I was trying to update brunch and Chrome OS together but due to network issue Chrome OS update cannot finish.
Brunch updated successfully.
When I rebooted device it showed 
`rootc is currept by an unfinished system update brunch will not rebuilt untill the update is complete`
And then I stuck at chrome logo.
Is it possible to mount "my files" or "downloads" in Linux? so that I can get my files? or any other workaround to fix the situation?

sebanc, 2020-09-30:
User files are encrypted in chromeos so you cannot access them from linux.

Do you use the dual boot or single boot method ?

ameyjoshi3068, 2020-09-30:
Triple boot with windows Linux mint and Chrome OS

sebanc, 2020-09-30:
Ok there is no easy way to fix this but in linux mint you can:
1) create a loop device for a chromeos recovery image using "losetup"
2) create another loop device for brunch chromeos.img using "losetup"
3) Use "dd" to copy partition 3 from the recovery image to partition 5 of the brunch chromeos.img

ameyjoshi3068, 2020-09-30:
Ok I will download recovery image and try it
Thank very much you for reply😊

closed 2020-10-02 by ameyjoshi3068