sebanc/brunch #558: Camera service in macbook - 100% CPU

heliofreitasbr01, 2020-10-01:
I have a Macbook Pro 2012. I know that Facetime camera (pcie) not working with image recovery. But the task below is consuming 100% CPU with the latest stable brunch and testing version too. I am using Samus 85. It's not possible kill this task because it back again some seconds later.

/vendor/bin/hw/android.hardware.camera.provider@2.4-service

sebanc, 2020-10-01:
There is a currently a known issue with chromeos consuming 100% of one cpu but it is related to the playstore:
https://www.reddit.com/r/chromeos/comments/j0y6pk/new_google_chromeos_updates_cause_100_cpu_usage/

Could you post your /var/log/arc.log file ?

heliofreitasbr01, 2020-10-01:
Here is:

localhost /var/log # tail arc.log
2020-10-01T08:02:37.144098-03:00 WARNING arc-kmsg-logger[3744]: healthd: battery l=96 v=12481 t=34.2 h=1 st=5 c=0 fc=5589000 cc=357 chg=
2020-10-01T08:02:38.171772-03:00 WARNING arc-kmsg-logger[3744]: healthd: battery l=96 v=12480 t=34.2 h=1 st=5 c=0 fc=5589000 cc=357 chg=
2020-10-01T08:02:40.221780-03:00 WARNING arc-kmsg-logger[3744]: message repeated 2 times: [ healthd: battery l=96 v=12480 t=34.2 h=1 st=5 c=0 fc=5589000 cc=357 chg=]
2020-10-01T08:02:41.248281-03:00 WARNING arc-kmsg-logger[3744]: healthd: battery l=96 v=12480 t=34.3 h=1 st=5 c=0 fc=5589000 cc=357 chg=
2020-10-01T08:02:42.269935-03:00 WARNING arc-kmsg-logger[3744]: healthd: battery l=96 v=12481 t=34.3 h=1 st=5 c=0 fc=5589000 cc=357 chg=
2020-10-01T08:02:46.372891-03:00 WARNING arc-kmsg-logger[3744]: message repeated 4 times: [ healthd: battery l=96 v=12481 t=34.3 h=1 st=5 c=0 fc=5589000 cc=357 chg=]
2020-10-01T08:02:47.399885-03:00 WARNING arc-kmsg-logger[3744]: healthd: battery l=96 v=12482 t=34.3 h=1 st=5 c=0 fc=5589000 cc=357 chg=
2020-10-01T08:02:48.423885-03:00 WARNING arc-kmsg-logger[3744]: healthd: battery l=96 v=12481 t=34.3 h=1 st=5 c=0 fc=5589000 cc=357 chg=
2020-10-01T08:02:52.523727-03:00 WARNING arc-kmsg-logger[3744]: message repeated 4 times: [ healthd: battery l=96 v=12481 t=34.3 h=1 st=5 c=0 fc=5589000 cc=357 chg=]
2020-10-01T08:02:53.548778-03:00 WARNING arc-kmsg-logger[3744]: healthd: battery l=96 v=12480 t=34.3 h=1 st=5 c=0 fc=5589000 cc=357 chg=
localhost /var/log #

heliofreitasbr01, 2020-10-01:
Well, I found a workaround for this:
- Kill the process by chrosh (kill -9 PID for ...provider@2.4-service)
- Open the app camera imediately (before the process get up again)
* Camera will working fine in Android and Chrome browser apps too.
- Close camera. The process will be dead.

heliofreitasbr01, 2020-10-02:
I just see this report in log:

--------- beginning of crash
10-02 00:04:38.469   109   202 F libc    : Fatal signal 6 (SIGABRT), code -6 (SI_TKILL) in tid 202 (HwBinder:109_1), pid 109 (cameraserver)
10-02 00:04:38.487  2124  2124 I android.hardware.camera.provider@2.4-service: Camera provider Service is starting.
10-02 00:04:38.488  2124  2124 D vndksupport: Loading /vendor/lib/hw/android.hardware.camera.provider@2.4-impl.so from current namespace instead of sphal namespace.
10-02 00:04:38.515  2124  2124 I CameraHAL: ArcCameraHostHolder initializing
10-02 00:04:38.516   124   197 D ArcBridgeService: Remote connection for process 2124 established
10-02 00:04:38.517   124   143 I ArcBridgeService: ArcBridgeHostProxy connection for process 2124 established
10-02 00:04:38.517  2124  2127 I ArcBridgeSupport: Bridge ready. Host version 48, proxy version 48
10-02 00:04:38.518  2124  2124 I CameraHAL: ArcCameraHostHolder initialized
10-02 00:04:38.518  2124  2124 I CameraHAL: Initialize:136: ARC bridge mode
10-02 00:04:38.518  2124  2124 I CameraHAL: Initialize:153: ArcCameraDelegate initializing
10-02 00:04:38.529  2124  2127 I CameraHAL: OnCameraServiceReady:304: CameraService is ready
10-02 00:04:38.529  2124  2124 I CameraHAL: Initialize:155: ArcCameraDelegate initialized
10-02 00:04:38.529  2124  2127 I CameraHAL: operator():302: Bridge ready (version=0)
10-02 00:04:38.531  2124  2124 I CameraHAL: ~ArcCameraDelegate:159: ~ArcCameraDelegate
10-02 00:04:38.531  2124  2124 I CamProvider@2.4-impl: Loaded "V4L2 Camera HAL v1" camera module
10-02 00:04:38.531  2124  2124 I CamProvider@2.4-impl: setUpVendorTags: No vendor tags defined for this device.
10-02 00:04:38.531  2124  2124 I ServiceManagement: Removing namespace from process name android.hardware.camera.provider@2.4-service to provider@2.4-service.
10-02 00:04:38.532  2124  2124 I android.hardware.camera.provider@2.4-service: Registration complete for android.hardware.camera.provider@2.4::ICameraProvider/legacy/0.
10-02 00:04:38.532   109   207 I CameraProviderManager: Connecting to new camera provider: legacy/0, isRemote? 1
10-02 00:04:38.532   109   207 I CameraProviderManager: Enumerating new camera device: device@1.0/legacy/0
10-02 00:04:38.533  2124  2128 I CamDev@1.0-impl: Opening camera 0
10-02 00:04:38.533  2124  2128 I CameraHAL: camera_device_open:28: enter
10-02 00:04:38.533  2124  2128 I CameraHAL: Initialize:136: ARC bridge mode
10-02 00:04:38.533  2124  2128 I CameraHAL: Initialize:153: ArcCameraDelegate initializing
10-02 00:04:38.543  2124  2127 I CameraHAL: OnCameraServiceReady:304: CameraService is ready
10-02 00:04:38.543  2124  2128 I CameraHAL: Initialize:155: ArcCameraDelegate initialized
10-02 00:04:38.543  2124  2127 I CameraHAL: operator():302: Bridge ready (version=0)
10-02 00:04:38.556  2124  2128 I CamDev@1.0-impl: Closing camera 0
10-02 00:04:38.556  2124  2128 I CameraHAL: closeHwDevice:435: enter
10-02 00:04:38.558  2124  2128 I CameraHAL: ~ArcCameraDelegate:159: ~ArcCameraDelegate
10-02 00:04:38.558   109   207 I CameraProviderManager: Camera provider legacy/0 ready with 1 camera devices
10-02 00:04:38.558   109   207 E CameraProviderManager: filterLogicalCameraIdsLocked: Failed to getCameraCharacteristics for id 0
10-02 00:04:38.558   109   207 I CameraService: onDeviceStatusChanged: Status changed for cameraId=0, newStatus=1
10-02 00:04:38.558   109   207 I CameraService: onDeviceStatusChanged: Unknown camera ID 0, a new camera is added
10-02 00:04:38.558   109   207 E CameraProviderManager: filterLogicalCameraIdsLocked: Failed to getCameraCharacteristics for id 0
10-02 00:04:38.613  2135  2135 I crash_dump32: obtaining output fd from tombstoned, type: kDebuggerdTombstone
10-02 00:04:38.613   127   127 I /system/bin/tombstoned: received crash request for pid 202
10-02 00:04:38.613  2135  2135 I crash_dump32: performing dump of process 109 (target tid = 202)
10-02 00:04:38.615  2135  2135 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
10-02 00:04:38.615  2135  2135 F DEBUG   : Build fingerprint: 'google/samus/samus_cheets:9/R85-13310.93.0/6857068:user/release-keys'
10-02 00:04:38.615  2135  2135 F DEBUG   : Revision: '0'
10-02 00:04:38.615  2135  2135 F DEBUG   : ABI: 'x86'
10-02 00:04:38.615  2135  2135 F DEBUG   : pid: 109, tid: 202, name: HwBinder:109_1  >>> /system/bin/cameraserver <<<
10-02 00:04:38.615  2135  2135 F DEBUG   : signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
10-02 00:04:38.615  2135  2135 F DEBUG   : Abort message: 'Status.cpp:149] Failed HIDL return status not checked: Status(EX_TRANSACTION_FAILED): 'DEAD_OBJECT: ''
10-02 00:04:38.615  2135  2135 F DEBUG   :     eax 00000000  ebx 0000006d  ecx 000000ca  edx 00000006
10-02 00:04:38.615  2135  2135 F DEBUG   :     edi f3061754  esi f2a035f0
10-02 00:04:38.615  2135  2135 F DEBUG   :     ebp f2a03518  esp f2a03488  eip f5534c19
10-02 00:04:38.624  2135  2135 F DEBUG   : 
10-02 00:04:38.624  2135  2135 F DEBUG   : backtrace:
10-02 00:04:38.624  2135  2135 F DEBUG   :     #00 pc 00000c19  [vdso:f5534000] (__kernel_vsyscall+9)
10-02 00:04:38.624  2135  2135 F DEBUG   :     #01 pc 0001fcb8  /system/lib/libc.so (syscall+40)
10-02 00:04:38.624  2135  2135 F DEBUG   :     #02 pc 00032253  /system/lib/libc.so (abort+115)
10-02 00:04:38.624  2135  2135 F DEBUG   :     #03 pc 00006fe5  /system/lib/libbase.so (android::base::DefaultAborter(char const*)+37)
[... cut ...]
10-02 00:04:40.797  2158  2158 E CameraProviderManager: filterLogicalCameraIdsLocked: Failed to getCameraCharacteristics for id 0
10-02 00:04:40.798   137   234 W CameraService_proxy: Could not notify mediaserver, camera service not available.
10-02 00:04:40.798   137   234 I CameraService_proxy: Could not notify camera service of user switch, retrying...
10-02 00:04:41.359   137   137 W Looper  : Slow dispatch took 1695ms main h=com.android.server.notification.NotificationManagerService$WorkerHandler c=null m=3

sebanc, 2020-10-02:
Tbh, I don't really understand this issue or how your camera might work if it is pcie, could you also attach a full dmesg ?

heliofreitasbr01, 2020-10-02:
Thanks for your help., Sebanc. The camera is not pcie. Is USB 2.0 integrated. I was wrong. Well, may you tell me how I can get the full dmesg?

sebanc, 2020-10-02:
Ah ok makes sense, to get a dmesg run "dmesg > ~/Downloads/dmesg.txt" in crosh shell, then attcah dmesg.txt from your downloads folder.

heliofreitasbr01, 2020-10-02:
[dmesg.txt](https://github.com/sebanc/brunch/files/5317560/dmesg.txt)

heliofreitasbr01, 2020-10-02:
Here is, my friend.

sebanc, 2020-10-02:
Do you have the same issue if you remove "android_init_fix" ?

heliofreitasbr01, 2020-10-02:
I haven't tried that. I'll try

heliofreitasbr01, 2020-10-02:
I just did it. No effect. (sorry for print scrreen in portuguese)

[image]

heliofreitasbr01, 2020-10-02:
Maybe this bug can be related to this issue:

https://bugs.chromium.org/p/chromium/issues/detail?id=1063631&q=android.hardware.camera&can=2

sebanc, 2020-10-02:
It could indeed be related to this issue, thanks for pointing it out :)

As we have more control over chromeos, we might still be able to do something about it.

Could you try deleting the below files and then reboot ?
- /etc/init/camera.conf
- /etc/camera/camera_characteristics.conf
- /lib/udev/rules.d/50-camera.rules

Is the android process still broken ? Is the camera working in chromeos / in android ?

sebanc, 2020-10-02:
Your comment with the chromiumos bug reference appears to have been removed, not sure if I did that by accident, I was trying to add a 👍

heliofreitasbr01, 2020-10-02:
> It could indeed be related to this issue, thanks for pointing it out :)
> 
> As we have more control over chromeos, we might still be able to do something about it.
> 
> Could you try deleting the below files and then reboot ?
> 
> * /etc/init/camera.conf
> * /etc/camera/camera_characteristics.conf
> * /lib/udev/rules.d/50-camera.rules
> 
> Is the android process still broken ? Is the camera working in chromeos / in android ?
> 
> Afterwards, to recover your system, you can just press "e" when you see grub and add "rebuild" to options.

Hey! Everything works fine now. Camera is working and the service at 100% of CPU is solved. :)

heliofreitasbr01, 2020-10-02:
Thanks, Sebanc! Great job guy 👍

dooloon, 2020-12-06:
> 谢谢，塞班克！很棒的家伙👍

my macbook pro 15 2012 stuck at chrome OS LOGO  same macbook？