sebanc/brunch #559: 5ghz wifi not found in the Samus recovery with any brunch

heliofreitasbr01, 2020-10-01:
On my macbook pro 2012, using Samus recovery, ChromeOS does not recognize the 5.ghz wifi network (as in the MacOSx) only at 2.4ghz. And even at 2.4ghz the wifi speed never exceeds 14 megabytes, even though the network has 100 megabytes per second. I have tried with all brunch and Samus recovery images.

Crescendo-BLYAT, 2020-10-01:
have you tried other recoveries?
other than samus...

heliofreitasbr01, 2020-10-01:
Yes. Only Samus is compatible with this 3th generation CPU. :(

sebanc, 2020-10-01:
Do you use the "broadcom_wl" option ?

If not, could you post the output of "lsmod" in crosh shell ?

heliofreitasbr01, 2020-10-01:
Yes, "broadcom_wl" is enable. Here is the lsmod result:

Module                  Size  Used by
snd_seq_dummy          16384  0
snd_seq                69632  5 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
tcp_diag               16384  0
inet_diag              20480  1 tcp_diag
tun                    36864  0
nf_nat_tftp            16384  0
nf_conntrack_tftp      16384  1 nf_nat_tftp
nf_nat_ftp             16384  0
nf_conntrack_ftp       16384  1 nf_nat_ftp
xfrm6_mode_tunnel      16384  0
xfrm6_mode_transport    16384  0
xfrm4_mode_tunnel      16384  0
xfrm4_mode_transport    16384  0
esp6                   20480  0
ah6                    16384  0
ip6t_REJECT            16384  2
ip6t_ipv6header        16384  0
rfcomm                 57344  2
cmac                   16384  3
algif_hash             16384  1
algif_skcipher         16384  1
af_alg                 20480  6 algif_hash,algif_skcipher
uinput                 20480  0
ccm                    20480  6
btusb                  53248  0
btrtl                  16384  1 btusb
btbcm                  16384  1 btusb
btintel                24576  1 btusb
bluetooth             434176  34 btrtl,btintel,btbcm,btusb,rfcomm
ecdh_generic           20480  2 bluetooth
bcm5974                20480  0
intel_rapl             20480  0
x86_pkg_temp_thermal    16384  0
intel_powerclamp       16384  0
crct10dif_pclmul       16384  0
crc32_pclmul           16384  0
ghash_clmulni_intel    16384  0
pcbc                   16384  0
aesni_intel           200704  68015
joydev                 24576  0
aes_x86_64             20480  1 aesni_intel
crypto_simd            16384  1 aesni_intel
cryptd                 20480  34007 crypto_simd,ghash_clmulni_intel,aesni_intel
acpi_als               16384  0
applesmc               20480  0
sr_mod                 28672  0
kfifo_buf              16384  1 a
snd_hda_codec_hdmi     49152  1
snd_hda_codec_cirrus    20480  1
b43                   393216  0
snd_hda_codec_generic    69632  1 snd_hda_codec_cirrus
snd_hda_intel          36864  3
mac80211              774144  1 b43
snd_intel_nhlt         16384  1 snd_hda_intel
snd_hda_codec         110592  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_cirrus
snd_hda_core           65536  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_cirrus
cfg80211              589824  2 b43,mac80211
mei_me                 24576  0
tg3                   155648  0
snd_hwdep              16384  1 snd_hda_codec
ssb                    53248  1 b43
mei                    65536  1 mei_me
thunderbolt           122880  0
zram                   24576  1
firewire_ohci          40960  0
bcma                   49152  1 b43
hid_apple              16384  0
hid_appleir            16384  0
i915                 1466368  24
intel_gtt              20480  1 i915
video                  40960  2 apple_gmux,i915
drm_kms_helper        131072  1 i915
input_leds             16384  0
mac_hid                16384  0cpi_als
glue_helper            16384  1 aesni_intel
input_polldev          16384  1 applesmc
apple_gmux             16384  0
industrialio           61440  2 acpi_als,kfifo_buf
sbs                    16384  0
cdrom                  36864  1 sr_mod
apple_bl               16384  1 apple_gmux
sbshc                  16384  1 sbs
snd_hda_codec_hdmi     49152  1
snd_hda_codec_cirrus    20480  1
b43                   393216  0
snd_hda_codec_generic    69632  1 snd_hda_codec_cirrus
snd_hda_intel          36864  3
mac80211              774144  1 b43
snd_intel_nhlt         16384  1 snd_hda_intel
snd_hda_codec         110592  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_cirrus
snd_hda_core           65536  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_cirrus
cfg80211              589824  2 b43,mac80211
mei_me                 24576  0
tg3                   155648  0
snd_hwdep              16384  1 snd_hda_codec
ssb                    53248  1 b43
mei                    65536  1 mei_me
thunderbolt           122880  0
zram                   24576  1
firewire_ohci          40960  0
bcma                   49152  1 b43
hid_apple              16384  0
hid_appleir            16384  0
i915                 1466368  24
intel_gtt              20480  1 i915
video                  40960  2 apple_gmux,i915
drm_kms_helper        131072  1 i915
input_leds             16384  0
mac_hid                16384  0

heliofreitasbr01, 2020-10-02:
Sorry about that. Solved!!! I use dual boot. So I forgot that I need update grub from Linux after change anything in crosh too. :)

closed 2020-10-02 by heliofreitasbr01