sebanc/brunch #560: Dell Latitude e6410 freezes with 83-85 versions (samus recovery)

heliofreitasbr01, 2020-10-01:
My Dell Latitude e6410 can only run with Samus 81. No 83, 85 or even any 86 beta version working with it. The ChromeOS always freeze at open any browser tab. So I need to force a boot from button to get it again. But only until the login. It freezes again. I have tested it with all brunchs versions too.

Here are the specs for this model:
https://www.cnet.com/products/dell-latitude-e6410-14-1-core-i5-520m-windows-7-pro-2-gb-ram-250-gb-hdd-series/

sebanc, 2020-10-01:
Indeed I have had several reports of intel 1st gen devices having this issue since chromeos r83, my guess would be that it is related to userspace gpu drivers (mesa or libva).

Even though there is nothing much I can do about it, Google might still fix it at some point so I am not sure whether to remove intel 1st gen from compatible list.

heliofreitasbr01, 2020-10-01:
Right. Thank for your attention :)

closed 2020-10-01 by heliofreitasbr01