sebanc/brunch #561: Keyboard backlight auto turn on after idel for about 30 sec on ThinkPad T470

propotato123, 2020-10-01:
Installed the (eve) version of ChromeOS and keyboard backlight will auto turn on after about 30-sec idle. Using the latest version of brunch.

Dejwster, 2020-11-12:
Had the same problem on my Thinkpad Yoga 370. You have to press fn + alt + brightness control until it turns off. If you reboot you'll have to do it again.

Tash254-ke, 2020-12-23:
I have been the same issue on ThinkPad T460s. I have tried to press Fn+Alt+Backlight keyboard(Space bar in my case) and still doesn't work.

DawidOchowiak, 2020-12-23:
@Tash254-ke Instead of spacebar, press the brightness down key for your display. Fn+Alt+Brightness Down Key for display. It should turn off the keyboard backlight.

Tash254-ke, 2020-12-23:
Thank you so much. It worked, mine is F6 & F7. At least it'll now be easier on what is left of this battery.

DawidOchowiak, 2020-12-23:
@Tash254-ke No problem, glad I could help.