sebanc/brunch #562: Boot from SD Card

cdurau, 2020-10-03:
Hey guys,

I installed brunch to the sd card for my surface go 2.
When I try to boot I get the following error: "error: disk `,7' not found. You need to load the kernel first."

Booting to usb works just fine. Booting to an image on the sd card works, too, but Linux and Play Store are not working. Only with the full installation on usb.

My grub.cfg: 
`menuentry "ChromeOS (boot from disk image)" {
        rmmod tpm
        img_part=/dev/mmcblk1p1
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path options=iwlwifi_backport
        initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

set timeout=2

menuentry "ChromeOS" {
  regexp --set disk "(^.+)(,gpt)" $root
  linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug
  initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
}
`

Like I said, booting from the image works, booting to a full installation on sd card does not. 

The sd card is at "/dev/mmcblk1" and the internal storage at "/dev/mmcblk0".

Did somebody get it working to boot from a sd card or at least booting from hdd to an installation on sd card or external storage?

Thank you very much!

sebanc, 2020-10-10:
Hi,

A pre-requisite for brunch to boot from an sdcard is that the UEFI bios supports booting from sdcard (else grub does not even detect the device).

I don't know for Surface go 2 but my Surface go 1 cannot boot from an sdcard. I would actually like to have the ability to boot from sdcard too but only Microsoft can add this feature to the UEFI bios.

cdurau, 2020-10-10:
You're right, both surfaces are not able boot from SD, but I read that it's somehow possible to boot the Linux Kernel from HDD and load the OS from SD. At least for Ubuntu and other distros. I don't know if it's possible for Chrome OS.

sebanc, 2020-10-10:
In theory it is possible to boot that way:
- install the brunch efi folder from partition 12 on hdd
- Add a boot option for brunch in the uefi firmware
- copy the kernel and initramfs from partition 7 in the hdd efi folder
- modify the grub config to boot with the local kernel and initramfs

However, with that setup you cannot update brunch unless you re-perfom all those tasks.

cdurau, 2020-10-10:
I had to recover the bootloader a few times because of some experiments with it and just got it to work again. I stay with USB for now :-)

Thanks anyway for the great support!

closed 2020-10-10 by cdurau

Hmmmmmmmmmm, 2022-10-18:
> 

After 2 years I would like to give it a try
Is there a way for me to modify "grub config" though? Don't know any software can do that right now
And it would be super helpful if you can specify "the kernel" and "initramfs" too!

yeahman45, 2023-06-13:
> > In theory it is possible to boot that way:
> > 
> > * install the brunch efi folder from partition 12 on hdd
> > * Add a boot option for brunch in the uefi firmware
> > * copy the kernel and initramfs from partition 7 in the hdd efi folder
> > * modify the grub config to boot with the local kernel and initramfs
> > 
> > However, with that setup you cannot update brunch unless you re-perfom all those tasks.
> 
> I boot chrome OS on tf card successfully through this way, everything works smoothly on the small tf card(U3 V30 a2), just like running on a SSD.
> 
> But I can't open shell in the crosh,it seems the OS is not at developmode， and the speaker and camera on my yoga duet 2021 is not working, so I have to access the shell.
> 
> However, I have no idea about how to switch to the developmode... Could you help me, thank you very much!
> 
> Well, I found the problem, I ran the kernel derectly without the parameter in grub.cfg, emmm, how it works...

can you explain a bit more the steps to boot from sd card please?

Kissycat, 2023-06-15:
> can you explain a bit more the steps to boot from sd card please?

Install branch to tf card first, then cp files in root-c to any other partition, at last edit grub.cfg to boot chromeos kernel from this partition instead.

yeahman45, 2023-06-15:
Thank you for the reply.
"root-c to any other partition"
> you mean partition on the internal ssd;right?

"at last edit grub.cfg to boot chromeos kernel from this partition instead."
> do you have an example of the entry?

Kissycat, 2023-06-15:
> you mean partition on the internal ssd;right?

Yes, on ssd, even folds in windows partition.

> do you have an example of the entry? 

Just use the branch's grub config, and edit partition number.

yeahman45, 2024-08-31:
> > you mean partition on the internal ssd;right?
> 
> Yes, on ssd, even folds in windows partition.
> 
> > do you have an example of the entry?
> 
> Just use the branch's grub config, and edit partition number.

I gave up last time as I could not boot from sd card. I want to give it another shot.
I copy files in root-c to any folder on the drive on my ssd; any folder? or the EFI partition?

> Just use the branch's grub config, and edit partition number.

My grub generated by the brunch installer looks like this:
```
menuentry "ChromeOS" --class "brunch" {
	img_path=/tmp/brunch/r100/chromeos-100.img
	img_uuid=069d9###-27##-00##-1543-806###6e6963
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path \
			console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
	else
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
	fi
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

menuentry "ChromeOS (settings)" --class "brunch-settings" {
	img_path=/tmp/brunch/r100/chromeos-100.img
	img_uuid=069d9###-27##-00##-1543-806###6e6963
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1 \
		cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

I am not sure that to change...

Kissycat, 2024-09-01:
> I gave up last time as I could not boot from sd card. I want to give it another shot. I copy files in root-c to any folder on the drive on my ssd; any folder? or the EFI partition?

Somewhere that grub on your computer could access when system boot, any partition with correct filesystem (It is upto what drivers your grub have, EFI's fat is always fine) on your ssd is OK.

> > Just use the branch's grub config, and edit partition number.
> 
> My grub generated by the brunch installer looks like this:

Replace '(loop,7)' with **/your/new/path/**

> 
> ```
> menuentry "ChromeOS" --class "brunch" {
> 	img_path=/tmp/brunch/r100/chromeos-100.img
> 	img_uuid=069d9###-27##-00##-1543-806###6e6963
> 	search --no-floppy --set=root --file $img_path
> 	loopback loop $img_path
> 	source (loop,12)/efi/boot/settings.cfg
> 	if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
> 		linux **/your/new/path/**kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
> 			cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path \
> 			console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
> 	else
> 		linux **/your/new/path/**kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
> 			cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
> 	fi
> 	initrd **/your/new/path/**/lib/firmware/amd-ucode.img**/your/new/path/**/lib/firmware/intel-ucode.img **/your/new/path/**/initramfs.img
> }
> 
> menuentry "ChromeOS (settings)" --class "brunch-settings" {
> 	img_path=/tmp/brunch/r100/chromeos-100.img
> 	img_uuid=069d9###-27##-00##-1543-806###6e6963
> 	search --no-floppy --set=root --file $img_path
> 	loopback loop $img_path
> 	source (loop,12)/efi/boot/settings.cfg
> 	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1 \
> 		cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
> 	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
> }
> ```
> 
> I am not sure that to change...

yeahman45, 2024-09-01:
thx! it boots but it is stuck on brunch framework is loading. if it could not detect the sd card, it should have return an error right?
I will enable verbose and see.

yeahman45, 2024-09-01:
lol silly question, how to enable verbose in the grub entry?

xulqarnaen, 2025-01-23:
> Hi,
>
> A pre-requisite for brunch to boot from an sdcard is that the UEFI bios supports booting from sdcard (else grub does not even detect the device).
>
> I don't know for Surface go 2 but my Surface [...]

Hello

Facing the same issue for my usb, trying to install it on my intel  pc