sebanc/brunch #568: options=rtl8723de not working on the latest brunch framework

itshk16, 2020-10-06:
I updated my brunch framework today to **_"Brunch r85 k4.19 testing 20200919"_** and I found that my wifi was not working. So, I tried using the command **_"options=rtl8723de"_** but found that it doesn't work so i had to revert back to the last **_stable_** release.

sebanc, 2020-10-09:
That's weird, "options=rtl8723de" has been reported working by other users, could you attach a dmesg to this issue ?

itshk16, 2020-10-10:
> I updated my brunch framework today to **_"Brunch r85 k4.19 testing 20200919"_** and I found that my wifi was not working. So, I tried using the command **_"options=rtl8723de"_** but found that it doesn't work so i had to revert back to the last **_stable_** release.

I got it working now. I was using Grub2Win for handling the dual boot, so when i added the command to the custom code in the chromeos submenu configuration, it started working.

closed 2020-10-10 by itshk16