sebanc/brunch #569: Crosh Shell Command not found

keweminer, 2020-10-06:
I just updated to the latest 85 and I can not run the shell command at the crosh prompt. Anyone else seeing this?

bprendie, 2020-10-08:
I also ran into this same issue... Additionally, on my system I can no longer use ctrl-alt-f2 at the login screen either.

daemongloom, 2020-10-09:
Have you disabled developer mode accidentally? That's why you can be missing shell and ctrl-alt-f2.

keweminer, 2020-10-09:
I too can not crtl-alt-f2 to a login screen. ... Developer mode may have been deactivated, but all that I did was run the Chrome os update, which I've done dozens of times without incident.

sebanc, 2020-10-10:
Could you verify that you have "cros_debug" on the kernel command line in your grub config ?

To do that:
- Press "e" on boot when you see grub and check if it is there (after "cros_secure")
- if it is not present, add it manually, press F10 to boot, launch cros shell and edit your grub config file from there.

keweminer, 2020-10-11:
Found the issue "cros_debugi" instead of "cros_debug" Clearly an issue with my vim skills :D

keweminer, 2020-10-11:
It now accepts "shell" but, I still can not crtl-alt-f2 to a login screen.

keweminer, 2020-10-14:
I'll close this, since I don't really need alt-ctl-f2 to work.

closed 2020-10-14 by keweminer