sebanc/brunch #571: Google Play Store and Android apps are not working sometime.

tkharoontk, 2020-10-06:
Hello guys. I am using brunch 85 testing wersion with rammus 85 latest. After installing everything is working fine accept Android apps and play store. Sometimes it will work after restart. Sometimes it's not. Even manage android preferences option is not working at the time of the issue. Also I tried stable version of brunch and many recovery including rammus 83 sammus, etc. Any help?

cdurau, 2020-10-08:
Same here.

On Surface Go 2

Crescendo-BLYAT, 2020-10-09:
Please read my solution here: https://github.com/sebanc/brunch/issues/549
Apparently its fixed in the latest version v85.0.4183.133 (via OTA update, not avail on recovery).

cdurau, 2020-10-09:
> Please read my solution here: #549
> Apparently its fixed in the latest version v85.0.4183.133 (via OTA update, not avail on recovery).

Thanks! Worked for me 😊