sebanc/brunch #572: Google Play Store slows down Chrome OS

shivanshusingh2007, 2020-10-08:
Whenever I open Play Store or any other Android app the whole OS becomes slow and sometimes unresponsive. No matter how many times I reboot.
It may cause because I have only 2 GB RAM and the slow Intel Celeron N3530 Processor but all other items like chrome, app launcher, linux apps etc. are working fast.
I am using rammus recovery image version 83 and brunch version 83 stable.

Lucaacer, 2020-10-08:
It's 100% a ram issue, the CPU is not so bad, after all.

I have tested chrome OS on several devices, even with low end atom CPUs, and the culprit is ram.

Chrome OS + android + crostini works with a normal usage is fine with a 4 gigs ram, excellent with 8 and barely usable with 2 only.

sebanc, 2020-10-08:
I never tried chromeos with 2gb of ram but my suggestions would be:
- Try reseting google sync : In chromeos: Settings -> People -> Sync and Google service -> Review your synced data -> Reset sync
- Update to beta channel (a bug related to the playstore and generating cpu workload has been fixed there)
- Disable android media scanner

shivanshusingh2007, 2020-10-08:
@sebanc 
- I have Resetted sync

- I think switching to beta channel is not good because I already have performance issues(with Windows) and you have not yet
   released brunch 86 testing and there are no recovery image for it so it might be unstable. Should I switch to the beta?

- I don't know how to disable android media scanner in chrome os.

daemongloom, 2020-10-09:
> * I think switching to beta channel is not good because I already have performance issues(with Windows) and you have not
>   yet released brunch 86 testing and there are no recovery image for it so it might be unstable. Should I switch to the beta?

M86 will be released next week, so it's already stable enough. You can update without recovery image, that's not a problem.

shivanshusingh2007, 2020-10-09:
Today I updated from Version 83 to 85 and the play store is working fine (sometimes slower but it's fine for me) .
Thanks for your support. I am closing the issue. Thanks.

closed 2020-10-09 by shivanshusingh2007

Vaman223, 2023-01-10:
pls help me same issui brya recovery intel i5 12400  16gb 3200mhz ram but playstore slows down the whole computer pls help someone