sebanc/brunch #573: Can I change the display resolution(native resolution)

CiSJiK, 2020-10-09:
Greetings. I've used this project to work my laptop to boot into Chrome OS as installing to it. Almostly works fine. But my laptop display has 4K(3840x2160) as default resolution but it makes more graphical resource uses. So I tried `video=1920x1080` in grub linux command line(the parameter starts like `linux <kernel options>`) but it doesn't work anyway. How can I change the native display resolution? Any kind of method welcome.

Crescendo-BLYAT, 2020-10-13:
Go to Settings (cog in the clock menu) > Device > Displays > Display size (adjust the slider).

CiSJiK, 2020-10-13:
I didn't mean that. That functions means changing DPI instead of reducing the resolution.
In short, I want to use `video=` parameter in grub2 boot option menu. It doesn't work anyway.
PS: `vga=` parameter is replaced into `video=` in grub2 and I'm using that.(Dual boot enabled with ubuntu and default grub2 is based as ubuntu.

sebanc, 2020-10-13:
In theory, "video=1920x1080" parameter should work.

Could you post a dmesg while booted with this parameter ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder to this issue)

CiSJiK, 2020-10-15:
[dmesg.txt](https://github.com/sebanc/brunch/files/5383035/dmesg.txt)
Here is my dmesg.txt and will uploading grub.cfg to report mismatched help to understand?

sebanc, 2020-10-15:
Not sure why the kernel command line does not work...

You can try installing xrandr with chromebrew and using it to change the resolution ? (not sure if it works, I never tried that)

CiSJiK, 2020-10-16:
I don't know how do I install that. But I'm using chrome os 83 with brunch 83 stable. I heard Chrome os is not used x11. So I guess xrandr is not working on chrome os 83. Also chromebrew is not worked with sudo. So it may not affect to base window system.

CiSJiK, 2020-10-16:
Update: I've updated latest chrome os version(86) with lastest brunch build r86_4.19, but there is no changes about this solution.
Additionally, base chrome os recovery file is nami(version: 83) which is very compatible with my laptop, Lenovo YOGA 720-13ikb(81C3) with 512GB nvme ssd. Base grub is from Ubuntu 20.04 LTS and set grub boot entry which includes Windows 10 and Ubuntu and Chrome os(Copied and pasted in Ubuntu grub-customizer, changed Ubuntu's grub2 bootx64.efi to Chrome OS grub2 bootx64.efi cause of compatibility issue). 
Also, here is the result what you adviced. I installed xorg-server to use xrandr but I can't use that. Even though installed libxrandr, still can't use that command. What should I install file to use xrandr?

sebanc, 2020-10-16:
Could you try "video=inteldrmfb:1920x1080-32@60" ?

Also do you know if 1920x1080 is a resolution supported natively by your laptop screen (in the EDID) ?

CiSJiK, 2020-10-17:
It is supported. I've tested via both Ubuntu and Windows. Also default resolution is 3840x2160.

I tried that but it doesn't work anyway.

CiSJiK, 2020-10-19:
I've heard about that Chrome os is using frecon instead of x11. So I guess there is a solution of change resolution but I can't understand each of those files in frecon relative folder whose function it.

CiSJiK, 2020-10-26:
Okay I tried those parameter before noresume parameter, run modetest and there is the result.
`trying to open device 'i915'...done
Encoders:
id	crtc	type	possible crtcs	possible clones	
70	41	TMDS	0x00000007	0x00000001
77	0	TMDS	0x00000007	0x00000002
79	0	DPMST	0x00000007	0x00000004
80	0	DPMST	0x00000007	0x00000008
81	0	DPMST	0x00000007	0x00000010

Connectors:
id	encoder	status		name		size (mm)	modes	encoders
71	70	connected	eDP-1          	290x170		2	70
  modes:
	name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot)
  3840x2160 60 3840 3888 3920 4000 2160 2163 2168 2222 533300 flags: nhsync, nvsync; type: preferred, driver
  1920x1080 60 1920 2040 2248 2576 1080 1081 1084 1118 172780 flags: nhsync, pvsync; type: userdef
  props:
	1 EDID:
		flags: immutable blob
		blobs:

		value:
			00ffffffffffff0006af2b1100000000
			201a0104a51d117802ee95a3544c9926
			0f505400000001010101010101010101
			01010101010152d000a0f0703e803020
			350025a5100000180000000f00000000
			00000000000000000020000000fe0041
			554f0a202020202020202020000000fe
			00423133335a414e30312e31200a0004
	2 DPMS:
		flags: enum
		enums: On=0 Standby=1 Suspend=2 Off=3
		value: 0
	5 link-status:
		flags: enum
		enums: Good=0 Bad=1
		value: 0
	6 non-desktop:
		flags: immutable range
		values: 0 1
		value: 0
	19 CRTC_ID:
		flags: object
		value: 41
	75 Broadcast RGB:
		flags: enum
		enums: Automatic=0 Full=1 Limited 16:235=2
		value: 0
	76 scaling mode:
		flags: enum
		enums: Full=1 Center=2 Full aspect=3
		value: 3`
I've used video=1920x1080 not video=inteldrmfb1920x1080-32@60

daemongloom, 2020-10-27:
You can try something like that:
`video=eDP-1:1920x1080M@60`

Replace eDP-1 with your screen interface name. Usually it is eDP-1 or LVDS-1, but it could be different for your device.

CiSJiK, 2020-10-28:
> You can try something like that:
> `video=eDP-1:1920x1080M@60`
> 
> Replace eDP-1 with your screen interface name. Usually it is eDP-1 or LVDS-1, but it could be different for your device.

I tried but inteldrmfb force the resolution into 3840x2160 so I also tried
`video=eDP-1:1920x1080M@60e` 
but when
`fb0: switching to inteldrmfb from EFI vga`
shows fb is crashed.

CiSJiK, 2020-10-29:
Okay. The process is reached at half about the problem. I've read the official kernel documentation which is contained about framebuffer(fb). 
When I read dmesg, inteldrmfb is loaded after efifb loads. So I figure out that, using `video=efifb:width:1920,height:1080` instead of `video=1920x1080`. Then efifb recognize the resolution as 1920x1080, inteldrmfb also read as 1920x1080. 
But when framebuffer is changing into inteldrmfb, the recognition is disappeared and recognize as 3840x2160. So I tried to find the documentation which contained about inteldrmfb, but only developer api which I can find appears. 
Could someone tells about the boot parameter of inteldrmfb?

CiSJiK, 2020-11-12:
Well, I found the command, modetest which contain to set the display mode. I tried to run it anyway but it won't run cause of windows manager which contains drm application. So I want to run that command while in booting. is it possible to run that command via init process?

kingo132, 2020-12-12:
> drm.edid_firmware

Hi, did you make drm.edid_firmware work? I'm trying to customize edid too.

sebanc, 2020-12-13:
Custom EDID is currently only enabled in "brunch-unstable" repo, you should be able to use it by creating your custom EDID in "/lib/firmware/custom_EDID.bin" and adding "drm_kms_helper.edid_firmware=custom_EDID.bin" on the kernel command line.

kingo132, 2020-12-14:
> Custom EDID is currently only enabled in "brunch-unstable" repo, you should be able to use it by creating your custom EDID in "/lib/firmware/custom_EDID.bin" and adding "drm_kms_helper.edid_firmware=custom_EDID.bin" on the kernel command line.

Thanks, sebanc, I managed to inject edid by enabling the CONFIG_DRM_LOAD_EDID_FIRMWARE option in kernel and recompiled brunch from source code. Then follow your guide to modify grub.cfg and put edid.bin to /lib/firmware.
It works like a charm, thanks bro!

CiSJiK, 2020-12-14:
> Thanks, sebanc, I managed to inject edid by enabling the CONFIG_DRM_LOAD_EDID_FIRMWARE option in kernel and recompiled brunch from source code. Then follow your guide to modify grub.cfg and put [...]

I even tried that method but he mentioned that is working on unstable repo. Thanks for trying and there is the conclusion.

Using video= parameter is not work with current version. Even trying to force the resolution it makes undisplayable.
Maybe EDID selector will be helpful to choose the resolution or customize the source code to choose EDID manually.

Thanks everyone to give the solution globally and I'm closing this issue. Have good day with this awesome os!

CiSJiK, 2020-12-22:
I'm reopening this issue cause it seems to not work to me as well.

First of all, I've used [https://github.com/akatrevorjay/edid-generator](edid-generator) to make edid.bin and followed the commented method except kernel changes cause it already done in current version. But when it boot, there is no signal to my laptop display. Other function works fine.

What should I have to do?

kingo132, 2020-12-22:
> I'm reopening this issue cause it seems to not work to me as well.
>
> First of all, I've used [edid-generator](https://github.com/akatrevorjay/edid-generator) to make edid.bin and followed the [...]

You should use brunch version 87 or later, and then
1. add drm.edid_firmware=your_edid.bin to the kernel option in grub.cfg.
2. put your_edid.bin under /lib/firmware/.
Make sure your_edid.bin is valid with correct information in it.

CiSJiK, 2020-12-22:
I'm dual-booting with Ubuntu via grub which integrated in that os. So there is no ROOT-C partition. If your guess is right, I can put my edid file into /lib/firmware when os has been turned on. Also I managed to edit grub.cfg to load my edid firmware too. Thus it happened.

Then the last solution you mentioned is my edid is invaild to load it. How can I verify my edid is vaild? Also how can I create it?

kingo132, 2020-12-22:
> I'm dual-booting with Ubuntu via grub which integrated in that os. So there is no ROOT-C partition. If your guess is right, I can put my edid file into /lib/firmware when os has been turned on. Also [...]

You can dump your edid in windows by using edid-generator or in ubuntu by using edid-decode.
You can test both of the edids dumped from this two methods.
By the way, I'm using the option drm.edid_firmware=eDP-1:my_edid.bin, since I only need to inject the edid of my built-in display. You can try to specify this port name too (in my situation, the port name of my build-in display is eDP-1, you can find your port name in /proc under your igpu's pci device folder, I think).

CiSJiK, 2020-12-23:
Well even my panel is unable to customize edid firmware. I tried the my_edid.bin which is created by [AW EDID Editor](https://www.analogway.com/apac/products/software-tools/aw-edid-editor/) and my_edid_linux.bin by [wxEDID](https://sourceforge.net/projects/wxedid/)

To understand easily, I will call those file name instead of original name called 1920x1080
my_edid.bin: Created in windows through AW EDID Editor
my_edid_linux.bin: Created in ubuntu through wxEDID

my_edid.bin is invaild then it automatically uses default edid which integrated in my panel and my_edid_linux.bin is works but can't display, it only display as black. I got dmesg.txt to compare both binary files. Is there no solution for this?

dmesg log with using My_edid.bin
`[    0.000000] Command line: BOOT_IMAGE=(loop,7)/kernel boot=local drm.edid_firmware=1920x1080.bin noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug loop.max_part=16 img_part=/dev/nvme0n1p7 img_path=/chromeos.img console=vt.global_cursor_default=0 brunch_bootsplash=default
[    0.135086] Kernel command line: BOOT_IMAGE=(loop,7)/kernel boot=local drm.edid_firmware=1920x1080.bin noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug loop.max_part=16 img_part=/dev/nvme0n1p7 img_path=/chromeos.img console=vt.global_cursor_default=0 brunch_bootsplash=default
[    0.840115] drm_get_panel_orientation_quirk called with width=3840 height=2160
[   12.697043] fb0: switching to inteldrmfb from EFI VGA
[   12.698623] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   12.698626] [drm] Driver supports precise vblank timestamp query.
[   12.700594] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[   12.703825] [drm:edid_load] *ERROR* Size of EDID firmware "1920x1080.bin" is invalid (expected 128, got 256
[   12.706582] i915 0000:00:02.0: [drm] Panel advertises DPCD backlight support, but VBT disagrees. If your backlight controls don't work try booting with i915.enable_dpcd_backlight=1. If your machine needs this, please file a _new_ bug report on drm/i915, see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.
[   12.706616] drm_get_panel_orientation_quirk called with width=3840 height=2160
[   12.710742] [drm] Initialized i915 1.6.0 20200114 for 0000:00:02.0 on minor 0
[   12.790122] frecon[511]: Frecon using drm driver i915, version 1.6, date(20200114), desc(Intel Graphics) using atomic
[   14.233283] frecon[511]: Frecon using drm driver i915, version 1.6, date(20200114), desc(Intel Graphics) using atomic`

dmesg log with using my_edid_linux.bin
`[    0.000000] Command line: BOOT_IMAGE=(loop,7)/kernel boot=local drm.edid_firmware=1920x1080.bin noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug loop.max_part=16 img_part=/dev/nvme0n1p7 img_path=/chromeos.img console=vt.global_cursor_default=0 brunch_bootsplash=default
[    0.130256] Kernel command line: BOOT_IMAGE=(loop,7)/kernel boot=local drm.edid_firmware=1920x1080.bin noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug loop.max_part=16 img_part=/dev/nvme0n1p7 img_path=/chromeos.img console=vt.global_cursor_default=0 brunch_bootsplash=default
[    0.676683] drm_get_panel_orientation_quirk called with width=3840 height=2160
[   12.207338] fb0: switching to inteldrmfb from EFI VGA
[   12.208499] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   12.208501] [drm] Driver supports precise vblank timestamp query.
[   12.211151] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[   12.212923] [drm] Got external EDID base block and 0 extensions from "1920x1080.bin" for connector "eDP-1"
[   12.212936] i915 0000:00:02.0: [drm] Panel advertises DPCD backlight support, but VBT disagrees. If your backlight controls don't work try booting with i915.enable_dpcd_backlight=1. If your machine needs this, please file a _new_ bug report on drm/i915, see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.
[   12.212967] drm_get_panel_orientation_quirk called with width=1920 height=1080
[   12.220968] [drm] Initialized i915 1.6.0 20200114 for 0000:00:02.0 on minor 0
[   12.270777] frecon[510]: Frecon using drm driver i915, version 1.6, date(20200114), desc(Intel Graphics) using atomic
[   13.465667] frecon[510]: Frecon using drm driver i915, version 1.6, date(20200114), desc(Intel Graphics) using atomic
[   46.036716] frecon[510]: Frecon using drm driver i915, version 1.6, date(20200114), desc(Intel Graphics) using atomic`

CiSJiK, 2020-12-28:
Well my panel edid is informed that it is only supported as 4K. Other resolutions meant the emulated resolution, Original resolution is still as 4K.  This is my fault well. So I will close this issue. Sorry to all of contributer and sebanc to confusing this issue but thanks to make me understand to make the solution via all of method. I learned some information of displaying in linux through this issue.
Sorry and Thanks once again.

closed 2020-12-28 by CiSJiK