sebanc/brunch #574: Is there a scope of running this chrome OS in VMware

ABugNamedBeetle, 2020-10-11:
Things which are working:
I am successful in creating .iso and installing Chrome OS.
" chromeos_image_tester.sh" is also yielding ok

but I'm getting this error "brunch: ChromeOS was not found on any device"

[image]

sebanc, 2020-10-13:
Brunch does not work in virtualbox due to missing graphic drivers as explained in #303

closed 2020-10-13 by ABugNamedBeetle

Pikotaro, 2020-10-13:
it doesn't support iSCSI. you need to create a SATA drive instead. No iSCSI, NVMe or IDE support

ABugNamedBeetle, 2020-10-13:
thanks @Pikotaro for the help.
But its now stuck at  "ui main process terminated"(now i think this is gpu issue)
[image]

Pikotaro, 2020-10-14:
you can try disableing 3d acceleration... if you really want to run it in a virtual machine you could try installing ESXi on your PC and enable GPU passthrough.... But this might be not worth it