sebanc/brunch #575: Bootloop using reef recovery image

EstPC13, 2020-10-13:
I just tried to switch to reef from samus, given that my camera wasn’t working on the latter; however, I’m now stuck in a bootloop while booting ChromeOS, the command line goes by too fast to see anything.

Should I switch back to samus and just not use the camera, or is there anything else I can do? If so, is there any way to switch the recovery image without reflashing?

My CPU’s a Pentium NE56R, far from a new laptop; let me know if you need more info! Latest reef img and brunch version.

sebanc, 2020-10-13:
samus is the most compatible recovery image for camera support. However, chromeos only supports usb cameras while most Pentium devices have i2c or ipu ones.

I can confirm it if you post the output of "dmesg | grep uvcvideo" command in crosh shell.

EstPC13, 2020-10-13:
Interesting, my camera stopped working a couple updates ago with samus, and I saw someone here being recommended reef; it doesn't boot so I can' really access crosh, but I'll switch back to samus (though the camera probably won't be working) and report back.

sebanc, 2020-10-13:
Ah ok if your camera worked before it must be usb then,

Could you try the solution from #558 :
```
Could you try deleting the below files and then reboot ?

/etc/init/camera.conf
/etc/camera/camera_characteristics.conf
/lib/udev/rules.d/50-camera.rules
```

EstPC13, 2020-10-14:
I've already tried that, and can confirm it doesn't work; I'm sadly back on Windows until this is fixed, but I'll try fresh build from samus later (Windows is unberably slow)

sebanc, 2020-10-15:
I see no major change in the chromeos camera framework and I did not change brunch code related to the camera, it might be a temporary chromeos bug.

On which chromeos / brunch versions have you faced this issue ?

EstPC13, 2020-10-19:
ever since the dev 83 update hit; if it's a chromeos bug, it could be fixed with a reinstall. I'll close this, since I'll be on Windows, but I'll create a new one if the issue comes up again.

Thanks a lot for your support! It's probably some chromeos bug

closed 2020-10-19 by EstPC13