sebanc/brunch #578: is there any way I could run this thing  on VMware

varunkrishna123, 2020-10-14:
hi I want to run this thing on VMware because I couldn't install this when I try to install it show error like grub config MK something like that any help should be appreciated thank you note I try to dual boot Chrome OS 
using Lenovo v113 laptop with Intel i3 8th gen processor

sebanc, 2020-10-15:
Brunch does not work on vmware as there is no vmware graphics driver in chromeos. However, it is most likely possible to make brunch work on your laptop.

If you post the details of the issue you encountered, I can try to help.

motog2k14, 2020-10-16:
how about try merge some code from this in Brunch?

https://github.com/FydeOS/
chromium_os-vm-vmware

sebanc, 2020-10-16:
Unlike FydeOS and Cloudready, Brunch is not built from source so there is no proper way to add system components.

motog2k14, 2020-10-17:
@sebanc thank you for clarification and for Brunch project, I'm a long term fan, your dedication and work are appreciated and respected, add usefulness and fun for PC ecosystem.

my best regards

sebanc, 2020-10-17:
@motog2k14 thanks for the support :)

Jhone-Harry, 2022-11-14:
Hello, Installing ChromeOS on VMware Workstation Pro is what I want to do. There is a statement that "visual machines are not supported" on the GitHub downloads page. Is it possible to make work on VMware Workstation or VirtualBox?

Jhone-Harry, 2022-11-14:
Hello, Installing ChromeOS on VMware Workstation Pro is what I want to do. There is a statement that "visual machines are not supported" on the GitHub downloads page. Is it possible to make work on VMware Workstation or VirtualBox?

Please, if you can assist, I appreciate it.

tejasraman, 2022-11-15:
Hi all!

As mentioned in #303 you can't really install ChromeOS in a VM due to a system Mesa that's incompatible with most SVGA cards (VMware style virtual cards). Could you please close this issue?