sebanc/brunch #581: unable to boot after brunch update

ameyjoshi3068, 2020-10-15:
I recently updated to 86 both with brunch and chromeos. I first updated chromeos using ota update and restarted.
Then I updated brunch r86
But in the first boot, after rebuilding chrome OS rootfs it cannot go further command line.
In first boot what I noticed special:
rm sebanc brunch sources error no such file or directory
cp sebanc brunch sources error no such file or directory

I have attached video of second boot
[video of second boot](https://drive.google.com/file/d/1-4lYymbQRyUzfEZsIFDJyEdkRtH2ZBQE/view?usp=drivesdk)

sebanc, 2020-10-15:
It seems that for some reason you faced an issue when updating and partition 7 ROOTC is corrupt (according to the error message you saw, everything looks normal in the video except that apparently it does not boot further)

What you would need to do is to replace chromeos partition 7 with the rootc.img from the brunch archive.

I am trying to figure out what could have caused that but I am not sure yet.

sebanc, 2020-10-15:
Actually are you booting through a specific setup ?

It looks like the kernel booted by grub is not the same one as the modules

ameyjoshi3068, 2020-10-15:
thanks for your prompt reply
I am using grub 2 win (I like its interface) with 2 different chrome OS images. one is for stable and other is for dev channel.
with linuxmint and windows installed alongside
This problem has occured with stable one. other is booting fine
 
I will try replacing partition 7 and see if it fixes the problem

sebanc, 2020-10-15:
It would seem you have an issue with your grub config and you are booting the stable channel image with the kernel from the dev channel image.

ameyjoshi3068, 2020-10-15:
Oh it means it is problem with grub 
I will place chrome OS dev in linux grub and stable on windows grub
thanks for your valuable time

ameyjoshi3068, 2020-10-16:
It looks like that problem is not solved
I first deleted dev grub entry, updated grunb2win
Then I moved chrome OS dev to external storage
Then copied files from rootc.Img to partition 7 of chromeos.img
But same problem occurred this time I have video of what exactly happened at boot

https://drive.google.com/file/d/1-Oyn-pwFtvOZejbXZNqCp1xYYzzhZHMZ/view?usp=drivesdk

ameyjoshi3068, 2020-10-16:
my grub config
img_part=/dev/sda2
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug options=enable_updates,acpi_power_button,suspend_s3 loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img

ameyjoshi3068, 2020-10-16:
will it work if I only replace kernel from chromeos.img p7 with rootc.img from r86

ameyjoshi3068, 2020-10-16:
OK now I think the reasult does not worth the effort I am doing
I am probably going to reinstall chrome OS 
Thanks a lot

sebanc, 2020-10-16:
Ok I got it now. This is a limitation of brunch you cannot have 2 images with the same name and at the same place on 2 different filesystems (/ in your case).

Create one with a different name or place it at a different path.

ameyjoshi3068, 2020-10-16:
Oh that was problem
luckily I did not deleted my old chromeos.img and  now going to restore it
thanks for your help

closed 2020-10-16 by ameyjoshi3068