sebanc/brunch #584: RTL8821ce

MartyMacFly84, 2020-10-17:
Hi. 
Sorry if the same topic is already on. I used search engine and could not find anything.
My problem is that ive got a laptop with rtl8821ce wifi adapter and installed chrome OS with brunch. Wifi doesnt work.
 Used latest versions of both ChromeOS and Brunch
I know that you can somehow ad options to use rtl8821 into grub config. But here is my problem.
Im a complete noob in linux and chrome OS. 
Can someone please tell me what should i do and where exactly i should put those option cos im now losing a plot with it.

Thank you.

sebanc, 2020-10-17:
Have you performed a dual boot install or single boot ?

MartyMacFly84, 2020-10-17:
Single. Just chromeOs

sebanc, 2020-10-17:
press CTRL+ALT+F2 when you see the welcome screen, login as "root" run "edit-grub-config" and add "options=rtl8821ce" after "cros_debug"

MartyMacFly84, 2020-10-17:
Ok. Now. It might be a silly question but how do i acctually login as root.
It might be obvious for everyone but as i said in firts post im a complete noob.

sebanc, 2020-10-17:
Type "root" when you see the login prompt

MartyMacFly84, 2020-10-17:
right. so unfortunately CTRL+ALT+F2 doesnt work no matter what ill try. Any suggestions?

sebanc, 2020-10-17:
Ok then Press "e" quickly when you see grub, add a space and "options=rtl8821ce" after "cros_debug", login to chromeos, then:
Press CTRL+ALT+T
Type "shell" at the prompt
Type "sudo edit-grub-config"
Add "options=rtl8821ce" after cros_debug the same way as previously
CTRL+X to save

MartyMacFly84, 2020-10-17:
Thanks for the help but still no luck. Ive got a strange feeling that the internal wifi is broken. 
Is there anyway to get usb wifi adapter to work with it instead? Ive got tplink archer 2tu nano?

sebanc, 2020-10-18:
Unfortunately the "tplink archer 2tu nano" needs a specific driver and I cannot support all usb adaptors with specific drivers as there are too many. (anyway it would also need an option to be set...)

If you can connect to wifi in windows, the internal wifi module is probably not broken.

Are you sure your device uses rtl8821ce ? what's your laptop model ?

MartyMacFly84, 2020-10-18:
Thats the trouble i cant get it to work in windows either. Its a cheap "chuwi herobook pro 14.1" notebook. Im not 100% sure that this is rtl8821ce.
Ill have a look on the chipset when get back home.

sebanc, 2020-10-18:
It would seem that your device has rtl8723bu which is also available as an option.

Could you try the previous procedure with "options=rtl8723bu" ?

MartyMacFly84, 2020-10-18:
Nope. Still no luck. I honestly think that this wifi radio is broken. Surprisingly its a dual bluetooth and wifi chipset on this machine and bluetooth works absolutely fine. 
What wifi adapter should i buy to get it to work with this chromeos build?

sebanc, 2020-10-18:
I don't use wifi usb dongles so I don't really know which one you should buy but it would need to be compatible with kernel 4.19

Otherwise I can implement your usb dongle driver but it will take some time.

MartyMacFly84, 2020-10-21:
Hi. Sorry for late reply. If you could possibly do that that would be awsome.

sebanc, 2020-11-07:
I have added you to brunch-testing branch (you should have received a github invite), the release there contains the driver for your usb wifi adapter.

You will have to press "e" when you see grub and add "options=rtl8812au" on the kernel command line (after "cros_debug") for it to be enabled.

MartyMacFly84, 2020-11-07:
Hi. Thanks very much. Can you send the invite again as im getting 404 error

sebanc, 2020-11-07:
I cannot add you again, you have to click the validation link in the mail invite before you can access the repo.

MartyMacFly84, 2020-11-07:
Got it :) cheers. Will test it tomorrow.

ScardracS, 2020-12-19:
Hi there! I have the rtl8821ce chipset on my pc... Is it now fixed?

Funkydog-nl, 2021-01-06:
This one worked for me on my Lenovo V145-AST15

from @sebanc 
"Ok then Press "e" quickly when you see grub, add a space and "options=rtl8821ce" after "cros_debug", press F10 to boot and log in chromeos, then:
Press CTRL+ALT+T
Type "shell" at the prompt
Type "sudo edit-grub-config"
Add "options=rtl8821ce" after cros_debug the same way as previously
CTRL+X to save"

ScardracS, 2021-01-06:
> This one worked for me on my Lenovo V145-AST15
>
> from @sebanc
> "Ok then Press "e" quickly when you see grub, add a space and "options=rtl8821ce" after "cros_debug", press F10 to boot and log in [...]

Sadly I have an amd cpu and it doesn't start on my PC

Funkydog-nl, 2021-01-06:
Me too AMD A9

ScardracS, 2021-01-10:
> Me too AMD A9
> 
> > This one worked for me on my Lenovo V145-AST15
> >
> > from @sebanc <https://github.com/sebanc>
> > "Ok then Press "e" quickly when you see grub, add a space and
> > "options=rtl8821ce" after "cros_debug", press F10 to boot and log in
> > chromeos, then:
> > Press CTRL+ALT+T
> > Type "shell" at the prompt
> > Type "sudo edit-grub-config"
> > Add "options=rtl8821ce" after cros_debug the same way as previously
> > CTRL+X to save"
> >
> > Sadly I have an amd cpu and it doesn't start on my PC
> >
> > <https://github.com/sebanc/brunch/issues/584#issuecomment-755198779>, or
> > unsubscribe

Which image do you use?

Funkydog-nl, 2021-01-10:
I used grunt

CDLLY, 2021-01-21:
> I cannot add you again, you have to click the validation link in the mail invite before you can access the repo.

Hi,
I saw that the unstable brunch released a version with the driver of rtl8821cu added two days ago. Does this mean that we can use rtl8821cu on the stable version soon?
Meybe I should not ask it here.⊙﹏⊙

jmpxy, 2021-01-23:
> I used grunt

Have you tried the "zork" recovery image? If you have an AMD Ryzen CPU, this may work for the lower-end versions.

acerivero, 2021-05-24:
just installed chrome os on my laptop half an hour ago, i noticed wifi is not working. knowing the wifi adapter of my laptop is a big plus, this really works on my HP laptop @sebanc  ..thank you very much..

Ok then Press "e" quickly when you see grub, add a space and "options=rtl8821ce" after "cros_debug", press F10 to boot and log in chromeos, then:
Press CTRL+ALT+T
Type "shell" at the prompt
Type "sudo edit-grub-config"
Add "options=rtl8821ce" after cros_debug the same way as previously
CTRL+X to save

The-Burning, 2021-07-13:
I have Quadruple boot with Arch,Zorin,Windows11 and chrome os with brunch. 
I have the same rtl8821ce adapter but the above method doesn't work for me 
```
┌──────────────────────────────────────────────────────────────┐
│ edit-grub-config does not work with dual boot installations.                                                                 │
│ In dual boot, you have to modify the grub config that was                                                                   │
│ created at the end of the image file installation process.                                                                     │
└──────────────────────────────────────────────────────────────┘

```

bybenjamin, 2022-07-20:
> karşılama ekranını gördüğünüzde CTRL+ALT+F2 tuşlarına basın, "root" olarak oturum açın, "edit-grub-config" komutunu çalıştırın ve "cros_debug"dan sonra "options=rtl8821ce" ekleyin

would it work to apply this for chrome os flex as well?