sebanc/brunch #585: No Bluetooth, Camera and 3.5mm jack

ijinesh, 2020-10-17:
Hi,
I am using this on Asus Transformer Mini T102ha. Everything is working fine, except:
- Bluetooth not working
- Camera not working
- 3,5mm jack not working

Is there a solution to any of these, particularly 3.5mm jack?
Thanks.
Jinesh

sebanc, 2020-10-17:
Hi,

Regarding bluetooth, you are probably affected by a common bug, bt devices are not displayed in the settings menu but you can connect through command line (bluetoothctl cf issue #67)

For the other 2 issues, this device camera will not work as chromeos only supports usb cameras (also I owned this device and as far as I remember there was no kernel driver for those cameras)

Regarding the headphone jack, I don't think it was ever supported in the linux kernel, I actually broke the speakers on mine while trying to fix headphones / microphone.

ijinesh, 2020-10-17:
> Hi,
>
> Regarding bluetooth, you are probably affected by a common bug, bt devices are not displayed in the settings menu but you can connect through command line (bluetoothctl cf issue #67)
>
> For the [...]

Hi Sebanc,
Thanks for your quick reply. 
Sorry for noobish question, how and where do I run the command line and where do I copy the BCM.mcd file... How do I get to the firmware folder?
Thanks.

sebanc, 2020-10-18:
I am not sure you need a specific bluetooth firmware, I was only reffering to the bluetoothctl commands:

```
sudo bluetoothctl
[bluetooth]# devices
[bluetooth]# scan on
[bluetooth]# pair YY:YY:YY:YY:YY:YY
```

I have a doubt now regarding the firmware, could you attach a dmesg to this issue (run "dmesg > ~/Downloads/dmesg.txt" in cros shell and attach dmesg.txt from your downloads folder).

ijinesh, 2020-11-07:
> I am not sure you need a specific bluetooth firmware, I was only reffering to the bluetoothctl commands:
>
> ```
> sudo bluetoothctl
> [bluetooth]# devices
> [bluetooth]# scan on
> [bluetooth]# pair [...]

Hi Sebanc,
Pls find Dmesg file attached here.
[dmesg.txt](https://github.com/sebanc/brunch/files/5504313/dmesg.txt)

ijinesh, 2020-11-07:
> Hi,
>
> Regarding bluetooth, you are probably affected by a common bug, bt devices are not displayed in the settings menu but you can connect through command line (bluetoothctl cf issue #67)
>
> For the [...]

I am not able to copy BCM.hcd to the firmware folder as i don't know how & where to access it (i am noob). Hence, even through commandline I am not able to connect to the bluetooth.
Any other way out?
Thanks.

sebanc, 2020-11-07:
Hi, thanks for the dmesg. As I first suspected, you should not need any specific firmware, just connecting manually your bluetooth device using the "bluetoothctl" command in crosh shell should be enough.

```
sudo bluetoothctl
[bluetooth]# devices
[bluetooth]# scan on
[bluetooth]# pair YY:YY:YY:YY:YY:YY
```

ijinesh, 2020-11-07:
> Hi, thanks for the dmesg. As I first suspected, you should not need any specific firmware, just connecting manually your bluetooth device using the "bluetoothctl" command in crosh shell should be [...]

Hi Sebanc,
I couldn't connect using command in crosh. What am i missing here?
Thanks.

sebanc, 2020-11-07:
After you have typed "sudo bluetoothctl" and then "devices" in crosh shell, what are the lines that appear ?

a-st, 2022-02-15:
I own the same device and can confirm that it works with the commands stated above. Also the bluetooth settings start to show devices nearby once the `bluetoothctl scan on`  command has been issued.

a-st, 2022-02-15:
For the headphone jack I manage to get it working with following command `amixer -c 0 cset name="Headphone Channel Switch" on`

a-st, 2022-02-15:
Alright, I got bluetooth and the headphone jack fully working now.

For bluetooth I created  `/etc/init/bluetooth.conf` with following content:

```
start on starting system-services

exec bluetoothctl scan on
```

For the headphone jack, I changed all occurences of `cset "name='Headphone Channel Switch' off"` to `cset "name='Headphone Channel Switch' on"` in following files

- `/usr/share/alsa/ucm/chtrt5645/HiFi.conf`
- `/usr/share/alsa/ucm/chtrt5645/HiFi.conf`
- `/usr/share/alsa/ucm2/chtrt5645/HiFi.conf`
- `/usr/share/alsa/ucm2/codecs/rt5645/SpeakerEnableSeq.conf`