sebanc/brunch #586: Stuck at ChromeOS LOGO

lokey0905, 2020-10-17:
i'm not good at english
I was Install ChromeOS on a USB flash drive via this tutorial: https://github.com/sebanc/brunch#install-chromeos-from-windows. But when login ChromeOS, it stuck at ChromeOS LOGO! Please help!
I use Brunch r86 k4.19 testing 20200919 and samus 
i'm use Toshiba C40-A
CPU I3-3110M
GPU AMD Radom HD 7600M Series
Ram 8GB

sebanc, 2020-10-17:
The problem is that the GPU "AMD Radom HD 7600M Series" is not compatible with chromeos.

Verify that the intel GPU is enabled in your bios.

megahorst, 2020-11-01:
Same here with A4-1200 APU with inbuild AMD Radeon HD 8180.
Is there any change to get this working?

chromer030, 2020-11-02:
And same here with Intel HD Graphics (Haswell) + AMD Radeon HD 4650

yanggis, 2020-11-07:
Hi, Im on the same boat, but my computer does not have any AMD graphics cards; It has only Intel Celeron J1900 with integrated graphics card. How to fix it?