sebanc/brunch #587: Smart Unlock Fail

ameyjoshi3068, 2020-10-18:
Brunch : r86 k4.19 stable 20201014   ChromeOS : Rammus Version 86.0.4240.77 (Official Build) (64-bit)
Device: Dell Inspiron 3647 pc with intel 4th gen processor and HD graphics 4GB RAM

Problem:  Smart lock does not work
Description: after bootup, first it shows Bluetooth is turned off, after a few seconds it starts scanning for my device (apparently).                    but cannot detect my device.
I have to enter the password to log in.

This feature stopped working since the r85 20200919 release
I was hoping this issue will get fixed in the latest release but it didn't.

This may help
output of lsmod
Module                  Size  Used by
tcp_diag               16384  0
inet_diag              20480  1 tcp_diag
snd_seq_dummy          16384  0
snd_seq                69632  5 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
tun                    36864  0
nf_nat_tftp            16384  0
nf_conntrack_tftp      16384  1 nf_nat_tftp
nf_nat_ftp             16384  0
nf_conntrack_ftp       16384  1 nf_nat_ftp
xfrm6_mode_tunnel      16384  0
xfrm6_mode_transport    16384  0
xfrm4_mode_tunnel      16384  0
xfrm4_mode_transport    16384  0
esp6                   20480  0
ah6                    16384  0
ip6t_REJECT            16384  2
ip6t_ipv6header        16384  0
crypto_simd            16384  0
glue_helper            16384  0
aes_x86_64             20480  93511
rfcomm                 57344  2
cmac                   16384  4
algif_hash             16384  1
algif_skcipher         16384  1
af_alg                 20480  6 algif_hash,algif_skcipher
uinput                 20480  0
ath3k                  20480  0
btusb                  53248  0
btrtl                  16384  1 btusb
btbcm                  16384  1 btusb
btintel                24576  1 btusb
bluetooth             438272  35 btrtl,btintel,btbcm,ath3k,btusb,rfcomm
ecdh_generic           20480  2 bluetooth
intel_rapl             20480  0
x86_pkg_temp_thermal    16384  0
intel_powerclamp       16384  0
crct10dif_pclmul       16384  0
crc32_pclmul           16384  0
sr_mod                 28672  0
ghash_clmulni_intel    16384  0
cdrom                  36864  1 sr_mod
cryptd                 20480  2 crypto_simd,ghash_clmulni_intel
dell_smm_hwmon         16384  0
ath9k                 155648  0
snd_hda_codec_realtek    98304  1
ath9k_common           32768  1 ath9k
snd_hda_codec_generic    69632  1 snd_hda_codec_realtek
ath9k_hw              450560  2 ath9k_common,ath9k
snd_hda_codec_hdmi     49152  1
mac80211              774144  1 ath9k
snd_hda_intel          36864  5
snd_intel_nhlt         16384  1 snd_hda_intel
ath                    32768  3 ath9k_common,ath9k,ath9k_hw
snd_hda_codec         110592  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
cfg80211              589824  4 ath9k_common,ath9k,ath,mac80211
snd_hda_core           65536  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
zram                   24576  1
r8169                  73728  0
snd_hwdep              16384  1 snd_hda_codec
mei_me                 24576  0
mei                    65536  1 mei_me
realtek                20480  0
i915                 1466368  22
intel_gtt              20480  1 i915
video                  40960  1 i915
drm_kms_helper        139264  1 i915
joydev                 24576  0
input_leds             16384  0
mac_hid                16384  0

sebanc, 2020-10-18:
Hi,

For now, I am not sure what could be causing this issue as I did not make changes that could impact bluetooth in a long time and both brunch r85 versions have the same kernel if I am not mistaken.

Could it be possible that a chromeos update broke this ?

Otherwise, the only difference I can think of would be a broken firmware (I download in each release the latest firmwares from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git). Do ath9k use firmware files ?

ameyjoshi3068, 2020-10-18:
OK I will try with different brunch versions and recovery images to find out what is wrong.
Any suggests are welcome

sebanc, 2020-10-18:
If ath9k uses a firmware that might be the first thing to check, what's the output of:
`dmesg | grep -i firmware`

ameyjoshi3068, 2020-10-18:
this is output

chronos@localhost / $ dmesg | grep -i firmware
[    0.189055] Spectre V2 : Enabling Restricted Speculation for firmware calls
[    0.332134] ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignore

sebanc, 2020-10-18:
ah ok there is apparently no firmware used so that's not it.

Also, have you tried setting up smartlock again ?

ameyjoshi3068, 2020-10-18:
> ah ok there is apparently no firmware used so that's not it.
> 
> Also, have you tried setting up smartlock again ?

Yes many times
In two versions chromeos stable 86 dev87 
with brunch 85 and 86

ameyjoshi3068, 2020-10-18:
also with different accounts

sebanc, 2020-10-18:
I don't use that feature usually but I managed to set it up and I just logged in with it. (I am not saying that you don't have an issue, just that it's not a general one)

Now that I see what smartlock looks like, what's the behavior when you are on the login screen ? Is the unlock icon saying smartlock is not setup ? does it have the circle animation while looking for a bluetooth device ?

ameyjoshi3068, 2020-10-18:
I created this issue because I didn't know whether anyone else have same problem or not.
I understand If the problem is only with me then this issue is actually meaningless. 

I will share the link of video I made of the issue for your record
https://drive.google.com/file/d/10MWRPi5RRLPy1bI2QuplgYnbq5BHSGpX/view?usp=drivesdk

Thank you very much.

sebanc, 2020-10-18:
Thanks for the video 👍 It's much clearer now.

Even if this bug does not affect everyone, it is still an issue.

Could you attach a dmesg to this issue ?

ameyjoshi3068, 2020-10-18:
[dmesg.txt](https://github.com/sebanc/brunch/files/5397279/dmesg.txt)

sebanc, 2020-10-18:
Well, I don't see weird things in the log, just that the driver seems to be taking too much time to start.

The main problem seems to be that the bluetooth signal is not strong enough, however google makes a lot of modifications to this wifi/bt driver as they use it in chromebooks so it is difficult to track the exact modification that's causing this issue.

If there is a recent enough brunch version which works fine, I would suggest rolling back to it for now.

Hopefully it this will be fixed in the next kernel update.

ameyjoshi3068, 2020-10-18:
Ok,
thank you very much

dubagunta-rohit, 2020-10-18:
for me its working fine.. but it take a min for the bluetooth to turn on and start seaching for my phone. i guess the wifi-bletooth card restarts during that 1min cause even the wifi reconnects. 

this wasnt the case before 85 update... i ave no idea if the chrome os 85 or the brunch 85 caused this

sebanc, 2020-10-18:
The main change that I can explain is that brunch r83 had a specific patch which was slowing down boot so that all drivers were loaded before boot. This patch has been reverted since then to have a faster boot.

This explains @dubagunta-rohit's point but not @ameyjoshi3068's one.

I can probably force load the bluetooth module early, I will try to implement it in the next release.

closed 2020-10-19 by ameyjoshi3068