sebanc/brunch #589: "rmod tpm" in grub

jhedfors, 2020-10-18:
I see that there is a new line "rmod tpm" that is added to the grub menu stanza that is presented at the completion of a fresh install. For me, with the inclusion of that line, I get an error when booting "unknown command rmod" before continuing.  What is this supposed to do?

sebanc, 2020-10-18:
most users dualboot using ubuntu / linux mint grub, however it suffers for more than 1 year of a known issue which prevents using loop devices unless "rmmod tpm" is set.

Therefore, that line was added by default due to the important number of issues related to this but you can safely remove that line if you don't need it.

jhedfors, 2020-10-18:
Thank you. After removing it, I no longer get the "unknown command" error for rmod and tpm, and boots normally.

jhedfors, 2020-10-18:
@sebanc After re-reading your comments, I am wondering is the command supposed to be "rmmod tpm" instead of "rmod tpm"? Maybe there is a typo in the script?

closed 2020-10-18 by jhedfors

sebanc, 2020-10-18:
Which grub are you using ? is it the one from brunch or another one like arch linux... ?