sebanc/brunch #590: Chrome OS not booting from USB

joni22u, 2020-10-18:
I followed every step and then restarted my laptop. After a while brunch framework started to extract rootfs and after that was finished there was just an _. Nothing was happening I left it like that for about an hour and still nothing. I'm using brunch r86 and chrome os recovey image samus.

sebanc, 2020-10-18:
what's your laptop model ?

joni22u, 2020-10-18:
It's hp laptop 15-ra0xx

sebanc, 2020-10-18:
I looked at the specs of your laptop and it seems like it should work.

However, I have no idea why it won't boot right now, could you take a short video of the boot process so that I can check if there is some message that could help me diagnose the issue ?

joni22u, 2020-10-19:
sebancsorry for bothering you but i tried it again and this time left my laptop on all night in the morning I was greeted by Chrome OS welcome screen. I'm gonna delete this issue, but before I do could you tell me how to enable the rtl8723de wifi card? I've read your instructions but I can't figure out how to do it. Send an example please.
Greets.

closed 2020-10-19 by joni22u