sebanc/brunch #591: Safemode/Disable Extensions? Enabled flags, now ChromeOS crashing....

jhedfors, 2020-10-18:
I enable some experimental flags, now chrome/chromeos crashing back to login screen.  Anyway to fix without powerwash/rebuild?

Tash254-ke, 2020-10-19:
Which flags did you enable that might have caused this?

WesBosch, 2020-10-20:
ChromeOS does actually have a safemode that activates if it detects too many crashes in a row, it will automatically turn off all flags in that scenario

jhedfors, 2020-10-21:
@WesBosch As i understand, and from my experience, that is not work for this scenario. Flags are not turned off and a powerwash is the only way to resolve.

Further discussion here;
https://bugs.chromium.org/p/chromium/issues/detail?id=660170

jhedfors, 2020-10-23:
@sebanc OK, well same thing happened again, only now it can't be explained away by enabling flags, as i have not touched the flags since I powerwashed a few days ago.  I had booted into Windows and or Linux Mint and back to Chrome OS.

When logging in, Chrome browser attempted to load, and had the "restore pages" prompt. Screen would not accept any input then went black after a few seconds, and it attempted to load again. This repeated a couple more times before returning to log-in screen.  I am using Mint now, and am not sure I want start over with a fresh installation every week.

sebanc, 2020-10-24:
The only things I can think of that could cause that are:
- corrupt chromeos synced data (might be due to flags notably)
- custom boot scripts not behaving correctly

I have the same chromeos install for 2 months now, also regularly booting ubuntu and occasionally windows and never faced this issue.

jhedfors, 2020-10-24:
> The only things I can think of that could cause that are:
>
> * corrupt chromeos synced data (might be due to flags notably)
> * custom boot scripts not behaving correctly
>
> I have the same chromeos [...]

I have no flags enabled (I have reset them to be sure), and have not custom boot scripts. Hopefully it does not happen again. Thanks.

closed 2020-10-29 by jhedfors