sebanc/brunch #592: ChromeOS was not found on any device

BoBoBo51, 2020-10-20:
I'm going to install chrome os in my vmware, but this happened, some one help me please
[image]

sebanc, 2020-10-20:
Sorry but Brunch does not work in virtualbox due to missing graphic drivers as explained in #303

BoBoBo51, 2020-10-20:
So it only works on real machines or what?

sebanc, 2020-10-20:
yes only on the computers matching the specs detailed in the readme

BoBoBo51, 2020-11-08:
Okay sebanc, now i installed it on a real machine but it still is the same, any answer?

sebanc, 2020-11-08:
Chromeos not found message ?

BoBoBo51, 2020-11-08:
Yes

BoBoBo51, 2020-11-08:
Same message

sebanc, 2020-11-08:
I will need a photo of the screen with chromeos not found message and your grub configuration

BoBoBo51, 2020-11-08:
Okay 😔

sebanc, 2020-11-08:
Also what's your laptop model ?

BoBoBo51, 2020-11-08:
no im doing on a pc with a intel pentium dual core
[image]

sebanc, 2020-11-08:
As mentioned in the readme brunch is not compatible with intel cpu before 1st generation "Nehalem" (refer to https://en.wikipedia.org/wiki/Intel_Core)

BoBoBo51, 2020-11-08:
So my laptop has 4th gen chip, will it work?

sebanc, 2020-11-08:
yes 4th gen should work

BoBoBo51, 2020-11-08:
ok let me try

BoBoBo51, 2020-11-26:
Nah this time 4th gen cpu but idont have the internet when enter linux mint. Should i turn off secureboot or what

BoBoBo51, 2020-11-26:
I5 4200U
4GB of ram
Sp uefi boot
I think this should work

BoBoBo51, 2020-11-26:
hey sebanc

sebanc, 2020-11-26:
It seems you are not that familiar with linux, you should probably follow the windows install procedure

BoBoBo51, 2020-11-27:
What?

sebanc, 2020-11-27:
Try to create the brunch usb flashdrive under windows, it might be easier:
https://github.com/sebanc/brunch#install-chromeos-on-a-usb-flash-drive--sd-card

closed 2026-04-07 by github-actions[bot]