sebanc/brunch #593: Intel ac9462 doesnt work?

Billy233, 2020-10-20:
Firstly, thanks the author for developing so good framework to run CHROMEOS.
I install the system successfully, but I cant turn on the wifi button!! My wireless adpater is intel ac9462.
I have already tried download dirvers ***.ucode from Intel official website and copy it to /lib/firmware(ctrl + alt + f2 enter the console)..
However, It doesnt work.......
I got another solution by google. That page tells me to compile the UBUNTU driver and install it.But the system doesnt include gcc or g++ compiler and no package managerment.

I will extremely appreciated it if you can help me solve the problem.

sebanc, 2020-10-20:
Hi, have you tried adding "options=iwlwifi_backport" to the kernel command line ? (after cros_debug)

Billy233, 2020-10-20:
.............cros_debug options=iwlwifi_backport. Right?
It seems still not working.

sebanc, 2020-10-20:
In theory it should work, if you use the dual boot, you have to modify the grub config you have created in linux.

Could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder to this issue)

Billy233, 2020-10-20:
I find something important from dmesg, i think:
[   19.416143] Bluetooth: rfcomm_init_ttys() RFCOMM TTY layer initialized
[   19.416151] Bluetooth: rfcomm_init_sockets() RFCOMM socket layer initialized
[   19.416170] Bluetooth: rfcomm_init() RFCOMM ver 1.11
[   19.609684] iwlwifi 0000:00:14.3: Failed to load firmware chunk!
[   19.609690] iwlwifi 0000:00:14.3: iwlwifi transaction failed, dumping registers
[   19.609693] iwlwifi 0000:00:14.3: iwlwifi device config registers:
[   19.609724] iwlwifi 0000:00:14.3: 00000000: 02f08086 00100406 02800000 00800000 c2218004 00000000 00000000 00000000
[   19.609727] iwlwifi 0000:00:14.3: 00000020: 00000000 00000000 00000000 42a48086 00000000 000000c8 00000000 000001ff
[   19.609730] iwlwifi 0000:00:14.3: iwlwifi device memory mapped registers:
[   19.609763] iwlwifi 0000:00:14.3: 00000000: 18489004 00000040 00000000 00000000 00000000 00000000 00000000 00000000
[   19.609766] iwlwifi 0000:00:14.3: 00000020: 00000011 0c040005 00000351 d55555d5 d55555d5 d55555d5 80008040 001f0040
[   19.609783] iwlwifi 0000:00:14.3: Could not load the [0] uCode section
[   19.609788] iwlwifi 0000:00:14.3: Failed to start INIT ucode: -110
[   19.622351] iwlwifi 0000:00:14.3: Failing on timeout while stopping DMA channel 8 [0x0bad1122]
[   19.633767] iwlwifi 0000:00:14.3: Failed to run INIT ucode: -110

sebanc, 2020-10-20:
According to your dmesg, you have not correctly enabled "options=iwlwifi_backport".

kernel command line from dmesg:
```
(loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug loop.max_part=16 img_part=/dev/nvme0n1p5 img_path=/chromos.img
```

Billy233, 2020-10-20:
just put "options=iwlwifi_backport" after "cros_debug"? I exactly do it
I am going to censor the kernel command, wait a minute

sebanc, 2020-10-20:
You have to add "options=iwlwifi_backport" to the grub config you have created in linux.

Don't forget to run "sudo update-grub" afterwards

Billy233, 2020-10-20:
wow, its fine now!!!!!!!
thank you very much!

sebanc, 2020-10-20:
No pb :)

techxaiver, 2020-10-24:
Thanks very much!!!!!!
I have fixed this problem by what you said