sebanc/brunch #594: cant sign in have to force shut down and switch on 2-3 times to log in and bluetooth dosent work

zainullahk, 2020-10-20:
when i enter the password it dosent log in i have to keep restarting till i can login and bluetooth dosent work at all

sebanc, 2020-10-21:
For the login issue, try reseting google sync ? In chromeos: Settings -> People -> Sync and Google service -> Review your synced data -> Reset sync

The bluetooth displayed in the menu but working from command line is a known issue

zainullahk, 2020-10-21:
where is firmware folder in chromeos>?

zainullahk, 2020-10-21:
when will it be fixed? as my keyboard is broken and i can mysetype a lot and my password is long and without bluetooth nearby sharing dosent work smart unlock dosent work

sebanc, 2020-10-21:
When someone who faces this issue debug it and submit a pull request.

I cannot reproduce this on my devices

zainullahk, 2020-10-22:
> When someone who faces this issue debug it and submit a pull request.
> 
> I cannot reproduce this on my devices

heres a video of the problem
https://drive.google.com/file/d/1QEp7jnPVlfI4wVxwP_fYyYAEcFgeHoRL/view?usp=sharing

ameyjoshi3068, 2020-10-23:
The same problem also happened to me
If you have chromeOS working fine previously and suddenly problem has arisen then see (1)
If you have this problem since the first boot then see (2)

(1) It was also in my case 
        I) Update brunch with the latest update (re-update)
        II)If it doesn't work use any Linux USB boot and do [this](https://github.com/sebanc/brunch/issues/520#issuecomment-690047625) or [this](https://github.com/sebanc/brunch/issues/520#issuecomment-690064337)
       III)if it doesn't work reinstall brunch

(2) I can't help you much with this 
       you can try using another user account or by guest (to try)
      using framework option for Bluetooth https://github.com/sebanc/brunch#framework-options

Shercoolcool, 2020-12-02:
Same issue with pixel 2013 dual boots with windows.I use John Lewis's firmware.