sebanc/brunch #597: My laptop is not boot with USB

yeounhojun0607, 2020-10-22:
Hello,
My laptop stuck on black screen when I turned on laptop with USB.
Of course I change boot order, and my laptop is Lenovo Thinkpad X201i (Dual core i3-380M).
(I used Samus recovery, and I tried Rammus too, but it doesn't work)

Can you help me?

sebanc, 2020-10-22:
Does your laptop shows text on the screen like it is trying to boot or just black screen after you select usb boot ?

yeounhojun0607, 2020-10-22:
It does try somthing, but nothing else.
It shows blink “_” like this;
https://imgur.com/a/vlz8oTN.

sebanc, 2020-10-22:
It seems like your laptop won't boot uefi, you will have to re-create the usb and use the mbr patch:
- Extract the brunch archive in a directory
- Extract the "mbr_support.tar.gz" patch found in the github master branch in the same directory (overwriting files when asked)
- Follow the rest of the instructions as usual

yeounhojun0607, 2020-10-23:
Hello, thanks for your reply.
My notebook now boot on chromeos, but it stuck on Setting Complete screen.
I pressed “Start” button, but nothing happened.
I tried Ctrl Alt T to go to shell, but it failed too.
Do you have any idea?

yeounhojun0607, 2020-10-23:
Hello, thank you for your reply.
Now I booted with USB.

HWV I STUCK AFTER I LOGGED IN WITH MY GOOGLE ACCOUT.

[image]

It just have loading for like 50 minutes and it failed.
I tried about 3 times, and I tried restart too.
Can you help me with this prob too?
Thx:)

sebanc, 2020-10-23:
Unfortunately I am not sure why this happens...

You can try:
- Login to your Google dashboard account
- Go to the "Chrome" application -> Manage synced data
- Click on "reinitialize synced data"

Then try again to log in chromeos

sebanc, 2020-10-23:
Forgot something, also you can try enabling/disabling the TPM in the bios