sebanc/brunch #598: (URGENT) GPU Issue

GaViNHaLo, 2020-10-22:
Hello I am using a intel uhd 620 gpu and I have used this method to put chrome os on my pc it works with the cpu but when I went to play roblox it ran very bad then I turned on performance stats in roblox and saw the gpu was not being used at all I have used the rummus from Cros site the cpu works but gpu doesn’t hopefully I have done something wrong and there is a fix to this Thank You!

sebanc, 2020-10-23:
Unfortunately, I don't have much control over the way the android container works. I can try to see if I can think of something but I would need some additional info:
- What is your laptop model ? Does it have a dual gpu ?
- Which brunch version have you installed ?

GaViNHaLo, 2020-10-23:
Hello Thank You For The Quick Response. 

My Laptop Model is this https://www.lenovo.com/us/en/outletus/laptops/lenovo/ideapad-flex-series/Lenovo-ideapad-FLEX-5-1570/p/81CA000YUS.
 
im not sure if it is dual gpu since it is intel.

this is the branch I used “r85-k4.19-testing-20200919” 

Thank you and I hope I did something wrong.

sebanc, 2020-10-24:
Not sure what your issue is, I installed it to give it a try on my Surface Go and the performance was ok (even though the gpu shows 0 which is most likely normal).

My suggestions:
- add "enforce_hyperthreading=1" to the kernel command line
- maybe try nami recovery image

GaViNHaLo, 2020-10-24:
Sorry If in bothering you by asking this but How exactly do you edit the kernel and where do you put the commands I want to add enforce_hyperthreading=1 and console= command to it

sebanc, 2020-10-25:
In single boot: run "sudo edit-grub-config" and add a space and "enforce_hyperthreading=1" after cros_debug.
In dual boot: add "enforce_hyperthreading=1" after cros_debug in the grub config you have created in linux, then run "sudo update-grub" in linux.

closed 2026-04-07 by github-actions[bot]