sebanc/brunch #600: Starting the virtual machine Error starting penguin container: 5 Launching vmshell failed: Error starting crostini for terminal: 5

keweminer, 2020-10-24:
Ran updates today and now linux will not start. I'm using the latest Brunch stable release and updated to Chrome OS 86.0.4240.112.

sebanc, 2020-10-24:
You have probably not re-created the grub config in a long time and are missing the CPU firmware, replace the initrd line of your grub config with:
```
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
```
 if you use dual boot
Or
```
initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
```
 if you use single boot

keweminer, 2020-10-24:
Opps, I mean the linux container within Chrome OS.

sebanc, 2020-10-24:
Could you reboot, attempt to install crostini and post the "/var/log/messages" file ?

sebanc, 2020-10-24:
Also what's your laptop model ?

keweminer, 2020-10-24:
I shut it down and walked away for a minute. Just rebooted and the thing works fine now. (I changed nothing and this wasn't the first reboot after the update either.) Thanks for your help though.

keweminer, 2020-10-24:
Dell 5558

closed 2020-10-25 by keweminer