sebanc/brunch #601: Android video screens not working.

COOORN, 2020-10-24:
Whenever there's a screen where a video is supposed to play(Youtube, Game Pass cloud streaming, etc.), I can't see it, but can only hear it. I tried enabling the alternative android container option but it doesn't work. Is there a workaround to this or an option I have to enable?

sebanc, 2020-10-25:
What's your laptop model ?

COOORN, 2020-10-25:
Dell Inspiron 5000 2 in 1. It has an intel core i7 with 8gb ram. My other old computer(Inspiron 3000 core i3) works, though.

sebanc, 2020-10-25:
Ok, I was expecting it would be a very old laptop but it is not the case...

I guess you used "rammus" recovery image, could you try "nami" and "hatch" ?

COOORN, 2020-10-25:
I don't want to break my working set up with my other computer, sorry. I'm on my old computer now so I'm good now. Should I close this issue, since I don't have a problem with this anymore but it still doesn't work?

sebanc, 2020-10-25:
Yes, if you are fine with it, it is probably better to close it.

closed 2020-10-25 by COOORN