sebanc/brunch #602: can i disable native chromeos updates after enabling

sar777-dev, 2020-10-25:
I want to ask (mainly from sebanc) that can we disable native chromeos updates after enabling it using "enable_updates" kernal command statement because after downloading chromeos recovery images it may contain issue which may google solve by pushing  OS fix updates but if it continuously be enabled may get upgrade to next chromeos version and if brunch version not get updated or available, it may get ustable.

Crescendo-BLYAT, 2020-10-25:
The OS update is not auto in brunchbook, you need to trigger it manually in the Settings > About Chrome OS.

sar777-dev, 2020-10-26:
I had figured it out how to stop updating, simply remove added kernal command line modifications.

closed 2020-10-26 by sar777-dev

ayushguptahere, 2023-05-13:
> 

Can you say me the procedure