sebanc/brunch #603: Help regarding technical terms for beginners

ameyjoshi3068, 2020-10-25:
I have listed a few technical terms and help regading them below
I request sebanc to pin this issue for anyone who might need it
                                                                             **Edit Grub Config**:

 there are two possible situations here 
(1)If you have only chrome OS installed you again have two easy options
             {a}quickly press e when you see grub 
             {b}open terminal(Ctrl+Alt+T) first enter `shell` then type `sudo edit-grub-config`
(2)If you are dual-triple booting then edit the grub config you have created

**Usage**:
              (1)**Add options**: Add "options=option1,option2,..." (without spaces). For example: booting with "options=enable_updates,advanced_als" will activate both options.
              (2) **Add kernel parameters**:  place `enforce_hyperthreading=1` or `console=`  after "cros_debug" and before "options=...." if any
like this: 
[image]

gjbnh1701, 2020-10-26:
Thanks for posting this...really helpful for us noobs. However when I enter "sudo edit-grub-config" into the shell I'm greeted by the response "command not found"...I've copy and pasted to ensure no typing errors....

sebanc, 2020-10-28:
Thanks ! it's pinned :) I am just going to change the title

smaka510, 2020-12-29:
I only have chrome OS installed. there is no Terminal included and the following command requires root access
sudo edit-grub-config

How is this accomplished?

ameyjoshi3068, 2020-12-30:
Chrome OS does have a Crosh shell.
You can launch it by `Ctrl+Alt+T` and don't forget to enter `shell`

smaka510, 2020-12-30:
I found the terminal and got into the grub.cfg then I added "options=ipts" at the end of the "chromeOS" linux line.
Then I pressed "ctrl+o" then enter to save. It says "[wrote 14 lines]".

Now I press "ctrl+x" to exit. Then I use the command sudo reboot. I am expecting the computer to start up with the IPTS touchscreen (surface pro) working. But I get nothing.

This works when I add the command temporarily. What am I doing wrong?

smaka510, 2020-12-31:
Friend,
I figured it out... the command I entered was **"options-ipts" but should have been "options=ipts"**. I am so used to powershell that I didnt even notice. EVERYTHING IS WORKING. wifi, touchscreen and all!!! thank you so much!

RickoDean, 2021-12-19:
Hello all, I wanted to let you know that in Chrome/Brunch 94, the grub-edit-config settings are different. I did not end up having to add the additional code as listed here. I had to review a few screens that had different settings in them. The settings above were there and there were some that's not mentioned here. I used the up/down arrow to select the suggested settings, click space to tick the box and hit enter to go to the next page. I think it was 4 altogether. I was also able to change my brunch grub splash screen to different prepackaged (dark) theme. after ticking auto updates and restarting, I was able to update to version Chrome 96. no issues, nothing broken. Kudos Sebanc

kpskrishnan, 2022-01-14:
> I found the terminal and got into the grub.cfg then I added "options=ipts" at the end of the "chromeOS" linux line. Then I pressed "ctrl+o" then enter to save. It says "[wrote 14 lines]".
> 
> Now I press "ctrl+x" to exit. Then I use the command sudo reboot. I am expecting the computer to start up with the IPTS touchscreen (surface pro) working. But I get nothing.
> 
> This works when I add the command temporarily. What am I doing wrong?
> 
> sudo update-grub is "not found"

How did you get into grub.cfg? When I type "sudo edit-grub-config" it gives me a list of options. With drivers names such as broadcom_wl or rtbh.

I'm trying to enable wifi and bluetooth. Please help me.

I installed Chrome OS on my HP Pavilion 2014.
Thanks in advance

kpskrishnan, 2022-01-14:
> Hello all, I wanted to let you know that in Chrome/Brunch 94, the grub-edit-config settings are different. I did not end up having to add the additional code as listed here. I had to review a few [...]

Did you also have wifi issues? And did it get solved with the menu? If yes, then please tell me the steps in the menu for the same.

Thanks in advance.

RickoDean, 2022-01-14:
Hello, I have no issues at all with WiFi I thought I did until I realized that my WiFi switch was set to off. My audio and mic works well. I'm even getting chrome updates ota. I'm on most recent version of chrome yet my brunch is latest stable build. My only gripe is the ultra sensitive trackpad. But Things do get odd when I switch to beta/dev channels. I run into WiFi/audio issues (then)

kpskrishnan, 2022-01-15:
> Hello, I have no issues at all with WiFi I thought I did until I realized that my WiFi switch was set to off. My audio and mic works well. I'm even getting chrome updates ota. I'm on most recent version of chrome yet my brunch is latest stable build. My only gripe is the ultra sensitive trackpad. But Things do get odd when I switch to beta/dev channels. I run into WiFi/audio issues (then)
> 
> *did you check your grub edit config menu to make sure your WiFi isn't disabled?

I think I'm checking it right, but can you tell me again how to check it?

kpskrishnan, 2022-01-15:
> Hello, I have no issues at all with WiFi I thought I did until I realized that my WiFi switch was set to off. My audio and mic works well. I'm even getting chrome updates ota. I'm on most recent [...]

The grub menu needs to be accessed with the linux on usb? Or in the bios? Sorry, I'm a noob.

mihirshah28186, 2022-01-18:
I have dual booted my lenovo G50-70 Windows 10 laptop with Chrome os brunch release 93. I want to update it to the latest version.
How did you get into grub.cfg? When I type "sudo edit-grub-config" it gives me a list of options. With drivers names such as broadcom_wl or rtbh, but I do not find the command line where I have to type = enable updates, then save the updated grub config & exit grub config. Please help, need yr guidance.
Thanks & Regards in advance.

RickoDean, 2022-01-18:
> I have dual booted my lenovo G50-70 Windows 10 laptop with Chrome os brunch release 93. I want to update it to the latest version.
>
> How did you get into grub.cfg? When I type "sudo edit-grub-config" [...]

Use up and down arrows to navigate, click space bar to activate option & then hit enter to go to next page. In this case only select auto update option & hit enter on the remaining pages. Keep going until it says hit enter or click any key to reboot. That is how it works. Good luck

mihirshah28186, 2022-01-19:
For brunch release  93 which Kernel version is to be selected. The are 5 options present.
Thank you.

RickoDean, 2022-01-19:
> For brunch release  93 which Kernel version is to be selected. The are 5 options present.
> Thank you.

I did not change that setting. I hit enter, skipping it. I only wanted to add auto update

novalkomik, 2022-03-03:
Hi, if I remember it correctly, this should also works for the Grub2Win custom code. I'm almost gone insane when trying to save these options I choose from the grub config on the Chrome OS side.

But the question is, what would happen if I choose to put `options=$options` instead of `options= (my preferred options)`? Is it going to save the grub config I made in the Chrome OS side?

rodneylg, 2022-09-05:
The ChromeOS settings option freezes at the press enter prompt.  Additionally, the "sudo edit-brunch-config" and the "sudo edit-grub-config" commands aren't saving the parameters.  A more detailed explanation of how to add boot parameters to grub would be helpful.  For example, how do I force the os to boot using kernel 1.15 using grub etc.

plavitec, 2023-11-28:
> @smaka512 Chrome OS does have a Crosh shell. You can launch it by `Ctrl+Alt+T` and don't forget to enter `shell`

hello , i've done that but it says once I type shell that this command does not exist ( ERROR - ERROR: unknown command: shell)