sebanc/brunch #606: Touchpad not working Polaroid 1400

lolpockete, 2020-10-26:
hello !
firstly thanks for bringing this awesome os on our pc !
secondly, i've just installed this os and tried to change the grub with the 2 configs, but my touchpad isn't working, only my usb mouse is working, i've tried to reinstall, to try with samus, etc, but nothing changed, i just losted a whole day.

vineethrp, 2020-10-27:
I have this issue too, on my Onn 13.3 2 in 1. Its a kernel issue I think. I tried ubuntu with a later kernel(5.4) and it works. I am trying to get 5.4 kernel compiled and see if it works. Unfortunately build.sh doesn't complete successfully for me. Would need to debug and see what happens

lolpockete, 2020-10-27:
It's really strange, personnaly i don't think i can do more as a noob

vineethrp, 2020-10-27:
I have verified that 5.4 chromeos kernel works perfectly with the HTIX5288 touchpad. @sebanc , would it be possible to add support for 5.4 kernel from here:
https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-5.4

I tried building r85 with 5.4 kernel, but the build fails in crew when it tries to install pixman and subsequently fails for swtpm as well. I don't think build failing is a kernel issue. Could you please have a look when you get a chance.

lolpockete, 2020-10-27:
ok i just found a new thing, to make it "work", you should put another finger on the left side and use the pad as "normal" with your other hand, same thing on crouton, if we can emulate the finger on the left side of the pad, it will solve the problem...
I've no idea how to do that sadly

lolpockete, 2020-10-28:
any help @sebanc please ? i'm completely lost really 😭

sebanc, 2020-10-28:
You should both have received a github invite for "brunch-next" branch with 5.4 kernel.

@lolpockete I hope it will help with your issue.

@vineethrp Have you read #305 and does your touchpad works on crouton ?

lolpockete, 2020-10-28:
oh thank you ! i'll try it !

> You should both have received a github invite for "brunch-next" branch with 5.4 kernel.
> 
> @lolpockete I hope it will help with your issue.
> 
> @vineethrp Have you read #305 and does your touchpad works on crouton ?

vineethrp, 2020-10-28:
@sebanc, I did not try crouton yet. Its a kernel issue as far as I can see. In 4.19, the i2c-hid spews lots of errors, but 5.4 doesn't have any. So looks like its a i2c-hid or touchpad driver issue and not a config issue..

My problem with 5.4 is that Playstore is not working. I assume I am missing some patches in 4.19 that enables google playstore. Trying to figure out in those lines.

Similarly I am also trying to figure out if I can isolate the commits that fixes this issue in 5.4 so that we can backport it to 4.19. Will keep you posted..

sebanc, 2020-10-28:
@vineethrp when you say playstore is not working with 5.4 kernel you mean "brunch-next" or your own kernel ?

Could you try crouton with kernel 5.4 ?

lolpockete, 2020-10-28:
@sebanc 
oh god !
it seems to be working better rn, thanks really !!!
play store seems to be working too for me.
Thanks again

vineethrp, 2020-10-28:
@sebanc , it was my build that was not working. I just got to trying brunch-next and everything works perfectly. My 5.4 source was taken from chromium and I guess the source that you have on brunch-next might have extra patches on it to support android.

Thanks again, this works perfectly.

Now that we have verified 5.4 kernel works for this touchpad(HTIX5288), would you like to see the results for crouton. I had verified that 4.19 linux also has issues with this touchpad. If you would like me to try crouton 5.4, I shall try and let you know later today.

sebanc, 2020-10-28:
@vineethrp Nice to see that it works. It was definetly a kernel issue, no need to try crouton.